pybricks/support #689: beta.pybricks.com fails to load

dlech, 2022-07-28:
**Describe the bug**
With the update today, https://beta.pybricks.com fails to load with the following error:

    Failed to load resource: the server responded with a status of 404 () main.01b44ba1.js:1  

This only happens if you have visited the site previously.

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
What did you expect to happen instead?

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

dlech, 2022-07-28:
Looks like the problem was inadvertently introduce in https://github.com/pybricks/pybricks-code/commit/9cdd10247721d2f5544ab3e8500b1fd34e99b7c9.

dlech, 2022-07-28:
This has been fixed in v3.2.0b3 (Pybricks Beta v2.0.0-beta.5).

But I'm going to leave the issue open for a while since people who have used older versions of Pybricks Beta will still experience this bug when upgrading to the fixed version (because the old buggy version is already on their computer).

laurensvalk, 2022-07-30:
Not sure it is the same, but maybe this was also happening on Pybricks Code. See https://bricks.stackexchange.com/questions/17234/pybricks-on-android-tablet/17322#17322

dlech, 2022-07-30:
This particular bug has only been released on https://beta.pybricks.com, not https://code.pybricks.com like mentioned in the stackexchange question. But the same workaround is probably still applicable.

dlech, 2022-11-18:
I think it has been long enough now that we can close the issue.

closed 2022-11-18 by dlech