pybricks/support #692: [Bug]

BatteryPoweredBricks, 2022-07-28:
I am having an issue where my Cityhubs refuse to power off.  After running my trains I press the button on the hub to stop the program (hub LED is then blinking blue).  Then I hold down the button to try to power it off but most often it just starts blinking again.  

I am currently using the Beta version of Pybricks but I had the same issue before using the Beta version.

Bellow is a YouTube link to see what I'm experiencing.
https://youtu.be/cPMWrLPp1WQ

dlech, 2022-07-28:
Thanks for the bug report. I've observed the same thing on Move hub before, but not on the City hub.

Maybe we need to enable [this hack](https://github.com/pybricks/pybricks-micropython/blob/ed6aa3e3352b029ce135806b0f8debd2a1e7894a/lib/pbio/drv/ioport/ioport_lpf2.c#L462-L464) for City hub too?

laurensvalk, 2022-07-29:
Hi @BatteryPoweredBricks :wave: - thanks for coming over from the forums  . It really helps :)

Does your code do anything with the button? If so, it would be great if you want to share your code here. 

Which motors and sensors are connected in the video? This might help us reproduce the issue.

Could you also run the following script after the bug happens, and share what you see?

```python
from pybricks.hubs import CityHub

hub = CityHub()

print("Voltage:", hub.battery.voltage())
print("Reset reason:", hub.system.reset_reason())
```

> but not on the City hub.

We did have this related issue on City Hub before: https://github.com/pybricks/support/issues/385

BatteryPoweredBricks, 2022-07-29:
It's the program that was shared in the forum, I've only changed some of the settings.  I'll attach it below.

Voltage: 7812
Reset reason: 0

Is the voltage low enough to cause issues?  It certainly drives the train just fine.  Is there a general rule for a minimum voltage to eliminate it as an factor when having issues?

I have 2 train motors attached to the hub.

[BlueTrain.zip](https://github.com/pybricks/support/files/9216690/BlueTrain.zip)

laurensvalk, 2022-07-29:
Thanks for sharing. The [reset reason](https://docs.pybricks.com/en/latest/hubs/cityhub.html#pybricks.hubs.CityHub.system.reset_reason) looks normal and the voltage is totally fine as well. The hub automatically shuts down at low voltage (5V), so I just wanted to rule that out. But this seems unrelated.

Does this also happen with smaller programs? For example like the small test snippet I just gave you.

Does it make a difference if you run the program as stand-alone (saved on the hub) or from the Pybricks Code app?

BatteryPoweredBricks, 2022-07-29:
I just tried with the program in stand-alone mode with the train program and the test program.  Tried without a program loaded before and after playing a program from the app.  Same result for all of them, just keeps blinking

dlech, 2022-08-01:
Can we make a simpler test case to reproduce the problem? With the given program, I get:

```
Init Timer 1
Init Timer 2
Traceback (most recent call last):
  File "main.py", line 317, in <module>
  File "main.py", line 255, in portcheck
OSError: [Errno 16] EBUSY: Device or resource busy
```

So I am not able to reproduce the issue.

laurensvalk, 2022-08-01:
> Can we make a simpler test case to reproduce the problem? 

See:
> I just tried with the program in stand-alone mode with the train program and the [test program](https://github.com/pybricks/support/issues/692#issuecomment-1198911722). Tried without a program loaded before and after playing a program from the app. Same result for all of them, just keeps blinking

dlech, 2022-08-01:
> Or maybe we need to switch the gpio to input mode instead of output low?

I tried this on the Move Hub and it seem to fix the issue (instead of the previous hack of leaving VCC on). But since I can't reproduce the issue on the City Hub, I will need someone who can to confirm using this firmware:

https://github.com/pybricks/pybricks-micropython/suites/7616058926/artifacts/316138802

BatteryPoweredBricks, 2022-08-01:
I can try it out, I'm not sure how to select the firmware file to download it to the hub though.

dlech, 2022-08-01:
https://pybricks.com/install/technic-boost-city/#installing-the-latest-build-advanced

Once you download the firmware .zip file, you can drag and drop it on the flash firmware button at https://code.pybricks.com.

(Note, you will need to use https://code.pybricks.com since the button was removed in the beta version).

BatteryPoweredBricks, 2022-08-01:
I tried it on Ubuntu with Chromium and didn't get an error but it didn't load from the zip.  In Windows / Chrome it says

"The firmware.metadata.py file contains missing or invalid entries.  Fix it then try again."

dlech, 2022-08-01:
Darn, I forgot the new firmware isn't compatible with the old Pybricks Code. There currently isn't a way to do it on Pybricks Beta. But if you have an Ubuntu machine, you can try this:

```
sudo apt update
sudo apt install pipx
pipx run pybricksdev flash ~/Downloads/cityhub-firmware.zip
```

BatteryPoweredBricks, 2022-08-01:
I knew this dual boot would come in handy, yeah that worked like a charm.  The hub shuts down without issue.  I replaced the main.py in the zip with the train program I'm using and that worked as well :).

dlech, 2022-08-01:
Great, thanks for testing!

closed 2022-08-01 by dlech

BatteryPoweredBricks, 2022-08-24:
Follow-up and a quick question.  It seems my issue has to do with having a motor on port b, without it the hub turns off as it should.  I plan to make a video soon that will feature this program and a dual motor setup, I was wondering when the firmware would be released with this fix implemented.  It's no rush but I'm trying to schedule my next few videos out and didn't want to have to tell people to use Linux :)

laurensvalk, 2022-08-24:
> didn't want to have to tell people to use Linux

> I forgot the new firmware isn't compatible with the old Pybricks Code. There currently isn't a way to do it on Pybricks Beta.

Maybe we could allow dragging it into the settings pane.