pybricks/support #693: [Bug] Cityhub will not download Pybricks firmware using BT dongle

BatteryPoweredBricks, 2022-07-28:
I am having issues with the initial firmware download to my Cityhubs.  I am using Chrome, Windows 10 and a Plugable USB Bluetooth 4.0 Low Energy Micro Adapter.  I am able to update firmware on a Technic hub with no issue.  I am given this error no matter if using stable or beta

"The hub was disconnected before flashing was completed. Restart the hub and try again."

A few days ago I saw these errors by clicking more info or something like that.  I'm unable to find them now, just trying to pass on as much info as I can.

"GATT operation failed for unknown reason."

"GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`."

dlech, 2022-07-28:
Did you disconnect all motors and sensors? Is the battery low?

dlech, 2022-07-28:
If the problem is really with this specific Bluetooth dongle and you want to dig deep, You can log Bluetooth packets to see what is going on. More info at https://github.com/pybricks/support/discussions/270.

BatteryPoweredBricks, 2022-07-28:
> Did you disconnect all motors and sensors? Is the battery low?

I am able to download firmware even with motors attached using my Laptop and its integrated BT.  Today I tried without the motors attached just to be sure.  I've tried with multiple hubs as well

BatteryPoweredBricks, 2022-07-28:
assuming I've done this correctly this is the output

[2022/07/28 18:12:30.124835] OnConnectionStatusChanged() status=1
[2022/07/28 18:12:30.030782] OnConnectionStatusChanged() status=0
[2022/07/28 18:12:26.610985] OnGattDiscoveryComplete() success=1
[2022/07/28 18:12:25.716941] OnConnectionStatusChanged() status=1
[2022/07/28 18:12:24.332289] StartGattDiscovery()
[2022/07/28 18:12:24.104944] No advertisement flags found.
[2022/07/28 18:12:17.250547] Web Bluetooth Device Chooser initiating Bluetooth discovery session

dlech, 2022-07-28:
That looks like the chrome device logs. I don't see anything useful there. Bluetooth packet capture is really the best way to see what is going on.

BatteryPoweredBricks, 2022-07-28:
Oops, sorry. I think I have it this time, Attaching wireshark dump.  

[Pybricks Cityhub BT.zip](https://github.com/pybricks/support/files/9214664/Pybricks.Cityhub.BT.zip)

dlech, 2022-07-29:
Thanks for taking the time to do this. It looks like Windows want to know the "Peripheral Preferred Connection Parameters", but we are possibly sending bad data from the hub.

[image]

I will look into this farther.

dlech, 2022-08-01:
[city-hub-fiirmware-update.zip](https://github.com/pybricks/support/files/9235877/city-hub-fiirmware-update.zip)

I have attached a Bluetooth packet capture of a successful firmware update from my Windows laptop. It continues after the malformed packet.

Since this problem happens during the firmware update, that means it is using the LEGO bootloader which can't be updated, so any software bugs can't be fixed.

I think some Bluetooth chips and Bluetooth software stacks may be more tolerant of the buggy bootloader. So I think the only thing that can be done here is to use a different device (Windows/Mac/Linux/Android/ChromeOS) or a different Bluetooth dongle to flash the firmware.

BatteryPoweredBricks, 2022-08-01:
For now I'm using an Intel NUC I had laying around, I happened to have an internal wireless adapter that was recent enough to support BT LE.  Works just fine in Linux as well as Windows 10.  I purchased an ASUS USB BT adapter on a recommendation that "should" work with the Cityhub, I'd like to have a good recommendation to give to people looking to use Pybricks with their trains.  Thanks for taking the time to look into it!

BatteryPoweredBricks, 2022-08-01:
Ok the ASUS adapter arrived, I had the same issue at first until manually downloading the latest driver from ASUS.  No problems after that.  So the problem seems to be specifically with the Plugable BT adapter when used with Cityhubs.  Plugable does not have a driver outside of the one from Windows update due to another Windows update causing incompatibility.  Happy to have some answers so I can share with others, again thanks for the help :)

BatteryPoweredBricks, 2022-08-02:
I spoke too soon, after that initial firmware download it stopped working and has the same error.  I'll keep testing and report anything I find

dlech, 2022-11-26:
Does it work any better with firmware v3.2.0b5 (or later) from https://beta.pybricks.com?

BatteryPoweredBricks, 2022-11-27:
Yes!  I was able to load a firmware multiple times without issue, although the first time didn't work (I may have held the button down too long?)

dlech, 2022-11-27:
Glad to hear it is working. We'll go ahead and close the issue, then.

closed 2022-11-27 by dlech