pybricks/support #696: Easy way to "open in Pybricks code"

dlech, 2022-08-01:
This feature has been requested by several users through various channels (and something similar in #173). 

Currently, to get code into Pybricks code you either have to download a `.py` file to your computer and then open the file in Pybricks code or copy and paste code into a new file in Pybricks code.

It would be nice if there was an easier way to do this with just one click. 

One solution could be to implement a URL handler. This is [current expermimental](https://web.dev/pwa-url-handler/) in Chromium-based browsers, so we could keep an eye on that.

laurensvalk, 2022-08-02:
Another request here:

[image]

afarago, 2023-05-01:
Great discussion here.
A possible storage mechanism mentioned in #1004 is version control systems like github / especially with the power of gist.

Checking out extensions like https://chrome.google.com/webstore/detail/copy-to-gist/lofgjnadojidbnajbnflkfnfmgiipfmc I think it could be a viable option to add "save to gist" / "update gist" functionality and support.
This would still keep all data and authoziation in the user space (browser) away from pybricks, yet would add a tremendous power.

gist seems to have a relatively easy REST interface
https://docs.github.com/en/rest/gists/gists?apiVersion=2022-11-28#create-a-gist

If you have some time to provide a theoretical feedback whether you'd be for or against this direction I could try spend the time to implement this functionality in the IDE and add a drafty PR.