pybricks/support #697: Hub subcomponents not listed in TOC in docs

dlech, 2022-08-04:
https://docs.pybricks.com/en/latest/hubs/primehub.html

I would like to be able to navigate directly to the battery section of the hub page without having to manually scroll or search for the word "battery".  The examples are listed in the table of contents, but the actual subcomponents are not:

[image]

dlech, 2022-10-21:
This is caused by using `.. rubric::` instead of a proper heading. We need to replace all uses of `.. rubric::`.

laurensvalk, 2022-10-21:
It’s been a while, but it certainly was intentional. I think regular headings broke the continuity of what is part of the class or something.

dlech, 2022-10-21:
Restructured text doesn't allow indented headings so using rubric was probably a hack around that. But we should be able to easily remove the indent since we are already using the fully qualified method names in automethod.

laurensvalk, 2022-10-21:
I think it made the grey method boxes blue inconsistently as they were no longer considered part of the class or something like that.

laurensvalk, 2022-11-22:
Since this is mainly nice to have, let's skip it for V3.2.