pybricks/support #698: DriveBase doc suggestions

dlech, 2022-08-05:
**Describe the issue**
What is wrong or unclear?

**Suggested improvement**
How can we fix it or make it clearer?

- [ ] `DriveBase` class initializer: mention that wheel diameter is often printed on the side of the LEGO tire (if it says A mm x B mm, which is diameter?) and pin holes in beams are approx 8mm center in case you don't have a ruler handy, you can use that. The intializer API doc should also link to the "Measuring and validating the robot dimensions" section, otherwise it is easy to miss.
- [ ] Methods that don't have a speed parameter should mention the `settings()` method as the way to set the speed.
- [ ] The `settings()` method should mention the default values.
- [ ] Methods should document possible exceptions. For example, you get a `ValueError` if the `angle` in `turn()` is too large, but what is the largest allowable angle?

laurensvalk, 2022-08-05:
- [ ] Explain that you can change the default turning direction by reversing the left and right motor arguments.
- [ ] Consider allowing sub-millimeter (floating point) dimensions. The low-level code implicitly supports this already anyway.
- [ ] Add code examples.

laurensvalk, 2022-12-06:
Fixed and/or deferred to separate issues as needed.

closed 2022-12-06 by laurensvalk