pybricks/support #703: Unfortunately Bluetooth issues are somewhat common due to the wide variety of Bluetooth hardware and operating systems available. But the good news is that there are some simple things you can do to really help us out. If you get a Bluetooth error in Pybricks Code, here are some things you can do to help diagnose the issue.

legotronic, 2022-08-13:
Unfortunately Bluetooth issues are somewhat common due to the wide variety of Bluetooth hardware and operating systems available. But the good news is that there are some simple things you can do to really help us out. If you get a Bluetooth error in Pybricks Code, here are some things you can do to help diagnose the issue.

# Common problems

## Hub turns off while flashing firmware

Don't hold the button too long. The hub status light will turn off for a second while the flash memory is being erased. Then it will start flashing again. This is when you should release the button.

## Error when starting to flash firmware

Don't release the button too early, otherwise the hub may turn off before flashing firmware can start. The hub status light will turn off for a second while the flash memory is being erased. Then it will start flashing again. This is when you should release the button.

## City hub freezes while flashing firmware

Be sure that all motors and sensors are disconnected from the hub before flashing the firmware.

## The firmware still won't flash

There are some known issues with the bootloaders on the hubs (especially the City hub + Windows). Many users have reported success using a different operating to flash the firmware (Mac/Linux/Android/ChromeOS). Once the firmware is flashed, you will likely be able to connect using the original computer to connect.

## Windows Bluetooth stack gets in bad state

We haven't determined why, but sometimes the Windows Bluetooth stack will get in a bad state and will refuse to connect, even after multiple tries. We have found that restarting Windows can sometimes fix problems like this.

## Linux prints output twice

Sometimes we have observed information in the terminal being printed twice after disconnecting and reconnecting a hub on Linux. When this happens, BlueZ needs to be restarted with `sudo systemctl restart bluetooth.service`.

## Errors after upgrading from firmware v3.0 to >= v3.1 on Linux

See https://github.com/pybricks/support/issues/374#issue-929412949 for Pybricks Code Error and https://github.com/pybricks/support/issues/374#issuecomment-873563561 for pybricksdev error examples.

BlueZ keeps a cache of the device's GATT attributes and returns the old values. Clear the cache manually like this:

    bluetoothctl remove XX:XX:XX:XX:XX:XX
    sudo rm "/var/lib/bluetooth/YY:YY:YY:YY:YY:YY/cache/XX:XX:XX:XX:XX:XX"

where `XX:XX:XX:XX:XX:XX` is the Bluetooth address of the hub and `YY:YY:YY:YY:YY:YY` is the Bluetooth address of the Bluetooth adapter on your computer.

# Developer Console

This can tell us if there was a problem with the code we wrote.

1. Press <kbd>F12</kbd> or <kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>I</kbd> (<kbd>Cmd</kbd>+<kbd>Shift</kbd>+<kbd>I</kbd> on macOS) to open the web browser developer tools. 
2. Click on the **Console** tab.
3. Look for any error listed.

[image]

# Chrome Device Log

This can give us more information if there was an operating-system specific problem.

1. Open a new tab or window in your web browser.
2. Type `chrome://device-log` in the address bar and press <kbd>Enter</kbd>.
3. Click **Refresh** if needed.
4. Change the log level to **Debug**.
5. Enable **Detailed Timestamps**
6. Filter for only **Bluetooth** messages.
7. Look for any errors or other useful information.

[image]

# Capturing Bluetooth Packets

For the really tough problems, it is useful to capture the actual Bluetooth packets being exchanged with the Bluetooth chip.

Please refer to https://bleak.readthedocs.io/en/latest/troubleshooting.html#capture-bluetooth-traffic for instructions on how to do this.

_Originally posted by @dlech in https://github.com/pybricks/support/discussions/270_

legotronic, 2022-08-13:
We do not connect to Bluethooth. 😣

On the hub is firmware V4.0.0.7.

We can connect to the VSC, but not to the pybricks app.

Have already tried on several computers - without success. 😒

What else can we do?

laurensvalk, 2022-08-13:
Which hub are you using?

The EV3 is only supported in vscode.

For SPIKE and MINDSTORMS robot inventor, see http://beta.pybricks.com/

legotronic, 2022-08-14:
> Which hub are you using?
> 
> The EV3 is only supported in vscode.
> 
> For SPIKE and MINDSTORMS robot inventor, see http://beta.pybricks.com/

We work with SPIKE - and the beta.pypricks of course.

the hub is listed in the Bluethooth overview (active) - but the pybricks app does not find any hardware.

[image]

It cranks endlessly - according to the picture

dlech, 2022-08-15:
Did you install the Pybricks firmware on the SPIKE hub first?

legotronic, 2022-08-15:
> 

We had first updated the firmware via the SpikePrime app - and then wanted to load the other firmware with the pyBrick app - but if we can't get a connection, then it's going to be difficult => chicken/egg problem.... 

Is/was our procedure wrong?

dlech, 2022-08-15:
You have to flash the firmware using USB before connecting with Bluetooth.

[image]

dlech, 2022-08-15:
You should have also seen this message when you cancel/close the Bluetooth search after not finding your hub.
[image]

Did this message not appear?

dlech, 2022-08-15:
It should appear when you cancel or otherwise close the "endless" scanning dialog.

You can use either method pictured to flash the firmware.

legotronic, 2022-08-15:
> It should appear when you cancel or otherwise close the "endless" scanning dialog.
> 
> You can use either method pictured to flash the firmware.

Ok - we try again...

legotronic, 2022-08-16:
so now it is also clear where the difference between the online and the offline version of the byBricks app is...  😆

legotronic, 2022-08-16:
one last question:

 - firmware-change byBricks => check
 - beta.bypricks.com             => check
 - pybricks offline                 => Upload works => Message "new firmware version is available  => Cancel
 
 - After F5 => Error-Message "ValueError: incompatible .mpy file" => we do something wrong?
 
 
from pybricks.ev3devices import Motor
from pybricks.parameters import Port
from pybricks.tools import DataLog, StopWatch, wait

# Create a data log file in the project folder on the EV3 Brick.
# * By default, the file name contains the current date and time, for example:
#   log_2020_02_13_10_07_44_431260.csv
# * You can optionally specify the titles of your data columns. For example,
#   if you want to record the motor angles at a given time, you could do:
data = DataLog('time', 'angle')

# Initialize a motor and make it move
wheel = Motor(Port.B)
wheel.run(500)

# Start a stopwatch to measure elapsed time
watch = StopWatch()

# Log the time and the motor angle 10 times
for i in range(10):
    # Read angle and time
    angle = wheel.angle()
    time = watch.time()

    # Each time you use the log() method, a new line with data is added to
    # the file. You can add as many values as you like.
    # In this example, we save the current time and motor angle:
    data.log(time, angle)

    # Wait some time so the motor can move a bit
    wait(100)

# You can now upload your file to your computer

legotronic, 2022-08-16:
And: VSC and pyBricks => it works...?

laurensvalk, 2022-08-17:
> pybricks offline => Upload works => Message "new firmware version is available => Cancel
> After F5 => Error-Message "ValueError: incompatible .mpy file" => we do something wrong?

This shouldn't happen if the firmware was installed using https://beta.pybricks.com/. How did you install it?

> Create a data log file in the project folder on the EV3 Brick.  => it does'nt work for SPIKE...?!

The `DataLog` class is not available on SPIKE. If you'd like to see that supported on SPIKE one day, you could ask about it in a new issue.

legotronic, 2022-08-17:
> > pybricks offline => Upload works => Message "new firmware version is available => Cancel
> > After F5 => Error-Message "ValueError: incompatible .mpy file" => we do something wrong?
> 
> This shouldn't happen if the firmware was installed using https://beta.pybricks.com/. How did you install it?
> 
> > Create a data log file in the project folder on the EV3 Brick.  => it does'nt work for SPIKE...?!
> 
> The `DataLog` class is not available on SPIKE. If you'd like to see that supported on SPIKE one day, you could ask about it in a new issue.
> 
> > And: VSC and pyBricks => it works...?
> 
> There are ways to do it, but we strongly recommend using https://beta.pybricks.com/

The installation of the firmware was done with https://beta.pybricks.com/ - therefore also surprised by the said message.

in fact the DataLog class is for the EV3 hardware - we overlooked that in the heat of the moment.

We would like to write log data from SPIKE - so will set a ticket accordingly. After all, we have no other way to do this.

And regarding VSC => we would like to work with versioning in the background - therefore VSC would be our preferred engineering suite. Or else pyCharme.

legotronic, 2022-08-17:
and: Are you also familiar with this link? https://bit.ly/3pn8Euy 😲

laurensvalk, 2022-08-23:
> ValueError: incompatible .mpy file

Does this keep happening, or did it only happen the very first time after installing the firmware?

Does it happen when you run a program via the editor, or only if you start the program with the button?

Harry-cheong, 2023-01-09:
Hi. Is there a way to run python files on spike prime via a command line? I have tried many methods to connect the spike prime to VSC but to no avail.

dlech, 2023-01-09:
You can run Pybricks programs on a SPIKE Prime that is running Pybricks firmware from a command line using `pybricksdev`. If you are using the official LEGO firmware on the SPIKE Prime, you will need something else.

closed 2023-01-09 by dlech