pybricks/support #710: [Question] Program start after power up on RI Hub?

johnscary-ev3, 2022-08-28:
My program used to run after I power cycled the RI hub and hit the hub button a second time after power up.
Using RI firmware "firmware-version": "v3.2.0b3-46-g37e69699" now and it does not run anymore after power up.
If I download and run my program it is fine. I can stop the program with a button hit and then restart the program ok with another button hit.
If I turn off the hub, then power on and hit button second time it does not start the program.
Was this function removed or changed?
I read through some of the comments here and seems like there has been some work done on this feature but not clear what the result of all that was.
Thanks.

laurensvalk, 2022-08-28:
Thanks for asking. The implementation was recently updated. It was done in several steps, where we removed the old method before we added the new one. This made the code review process a bit easier.

But you have coincidentally used the one version right in the middle of it :smile:

It should be fixed by just upgrading to the very latest version, where all functionality has been restored.

Let me know if it works. You might be the very first to try it. Thanks!

laurensvalk, 2022-08-28:
If you upgrade to the latest pybricksdev (a30) as well, you can even try multi file projects.

You run a script as usual. But it is now allowed to import another python file (from the same folder on your computer). That one will be automatically downloaded to the hub as well.

laurensvalk, 2022-08-28:
Also tagging @bertlindeman who might want to try this.

BertLindeman, 2022-08-28:
> Also tagging @BertLindeman who might want to try this.

Literally a wakeup call, Laurens.
Have to state that the new "install firmware" is **so** user friendly.

Not on topic but I first tried

- Technic hub at ```('technichub', '3.2.0b3', 'v3.2.0b3 on 2022-07-20')```

- City hub at ```('cityhub', '3.2.0b3', 'v3.2.0b3 on 2022-07-20')```

OK both before and after power-cycle.
And also a power-button press still runs the internal program after some other program has been run.
(Hmm, I hope this isn't too cryptic)

Had problem with install of pybricksdev on win11 ```No Python at 'c:\program files\python39\python.exe'```
Is there already an support-item for that or one that I can add this to?

Used pybricksdev to upgrade firmware on PrimeHub (Robot-Inventor color) to build 2016 (That is a sequence not a year)

How to load an internal program onto the PrimeHub?
It can not be done while flashing the firmware...

laurensvalk, 2022-08-28:
To try the features being discussed here, you’ll want to try the latest firmware from the [master branch](https://github.com/pybricks/pybricks-micropython/actions?query=is%3Asuccess+branch%3Amaster+workflow%3ABuild) and install it with Pybricksdev.

It will eventually be coming to Pybricks Beta as well to make it even easier.

BertLindeman, 2022-08-28:
> To try the features being discussed here, you’ll want to try the latest firmware from the [master [...]

Wasn't that the firmware I used - build 2016 from 2 days ago?
```('primehub', '3.2.0b3', 'v3.2.0b3-55-gc126dd6b on 2022-08-26')```

I miss a starting point to how to load a program to start it with the power button. As that is the starting post in this question.

Also looking at multi-file support (Starting with [pybricks/pybricks-micropython/115](https://github.com/pybricks/pybricks-micropython/pull/115) )

laurensvalk, 2022-08-28:
Sorry, I was reading the July dates from the other hubs. Yes, the one from two days ago is fine.

You can run a program with pybricks code or pybricksdev and it should be saved on the hub. Just start it again with the button.

Does this not work?

Multi file projects are only supported in pybricksdev right now.

BertLindeman, 2022-08-28:
> Sorry, I was reading the July dates from the other hubs. Yes, the one from two days ago is fine.
>
> You can run a program with pybricks code or pybricksdev and it should be saved on the hub. Just [...]

Thank you, Laurens.

It's so simple and yes it works.

For clarity:
Run a program via pybricksdev and then press the power button on the primehub and the program runs again.
The last program stays in the hub and can thus be re-run.

laurensvalk, 2022-08-28:
> It's so simple and yes it works.

:tada: 

And this should now work on Move Hub / City Hub / Technic hub too.

johnscary-ev3, 2022-08-28:
I loaded up firmware build -2016 and the run program after power cycle feature is working again.
Thanks.

Next, I will try the "multi file project" feature that you mentioned.
I read through your PR comments on this new feature.
It looks like it collects up all dependent files and sends them each time. Correct?
Is it possible to have it only download files that are missing in the hub or have been updated?
That would greatly reduce download times, when you just edit one file in a big multi file project for example.
Thanks.

laurensvalk, 2022-08-29:
> It looks like it collects up all dependent files and sends them each time. Correct?

Yes, that is how it works.

The files are currently just attached to each other like a tape archive, so updating just one depencency is not trivial.

But now that I am looking at it, there is actually a pretty easy way now to make download and run about 4x faster in a different way. I'll post about it in a separate issue.

johnscary-ev3, 2022-08-29:
OK thanks.
I noticed that the new set of software seems to be doing downloads almost 2x faster than before.
Also, I am looking forward to your future improvements on that.

laurensvalk, 2022-09-01:
The original question was resolved, so we can close this.

closed 2022-09-01 by laurensvalk

johnscary-ev3, 2022-09-02:
One last comment.
I broke my large project into multiple .py files and tried your new "multi file project" mode.
It works great without any problems using pybricksdev and VS Code.
Thanks for that new feature.

laurensvalk, 2022-09-02:
Thank you for testing, that is awesome!