pybricks/support #716: [Bug] Program can restart on cancel on Technic Hub

laurensvalk, 2022-08-30:
**Describe the bug**
On a Technic Hub, the user program sometimes restarts when you stop it with the button. I have not seen this on the other hubs, yet.

It's possible we've had this for a long time, but maybe it is more noticable now we support restarting programs with the button.

Maybe we can also reproduce this within a user script to rule out any download-and-run logic issue.

**To reproduce**
This is not specific to any program, but the following helps to visualize it. When canceling, the light becomes blue when you press, and sometimes it turns green again upon release, which shouldn't happen.

```python
from pybricks import version
from pybricks.hubs import ThisHub
from pybricks.parameters import Color
from pybricks.tools import wait
from pybricks import version

print(version)

print("Hello, world")
hub = ThisHub()
hub.light.on(Color.GREEN)

wait(5000)
```

laurensvalk, 2022-08-30:
> Maybe we can also reproduce this within a user script to rule out any download-and-run logic issue.

It's the button:

```python
from pybricks.hubs import TechnicHub
from pybricks.parameters import Button
from pybricks.tools import wait, StopWatch

hub = TechnicHub()
hub.system.set_stop_button(None)
watch = StopWatch()

# Print the time between button state changes
state = False
while True:
    now = bool(hub.button.pressed())
    if now != state:
        print(watch.time())
        watch.reset()
        state = now 
```
Continuously clicking the button gives a steady output and sometimes jitter like this:

```
272
319
310
255
301
295
256
303
265
262
276
291
200
246
267
273
257
263
0       <---- button noise
1       <---- button noise
1
0
249
243
1
0
0
1
265
243
0
1
779
234
```

BertLindeman, 2022-08-30:
Also seen on cityhub running ```('cityhub', '3.2.0b3', 'v3.2.0b3 on 2022-07-20')```

```python
from pybricks.hubs import ThisHub
# from pybricks.parameters import Button
from pybricks.tools import StopWatch
# from pybricks.tools import wait

hub = ThisHub()
hub.system.set_stop_button(None)
watch = StopWatch()

# Print the time between button state changes
state = False
while True:
    now = bool(hub.button.pressed())
    if now != state:
        print(watch.time())
        watch.reset()
        state = now

```
Console:
```
3020
269
730
234
6
22
360
232
1103
873
509
662
1297
675
0
1
```

BertLindeman, 2022-08-30:
Not seen on the movehub (not in a few minutes) on ```('movehub', '3.2.0b3', 'v3.2.0b3 on 2022-07-20')```

dlech, 2022-12-15:
I wasn't able to reproduce this on my Technic hub but was able to reproduce on my City hub. And only one time I saw a 4 ms time on the Essential hub (no 0. or 1 on that hub).

closed 2022-12-15 by dlech

laurensvalk, 2022-12-15:
Would an interrupt approach have less overhead than polling?

And/Or are there any ways to configure the pins differently (avoiding any software process) or wouldn't this help due to how they are wired?

dlech, 2022-12-15:
Technically, I suppose would be lower overhead, but polling every 10 milliseconds is probably hardly measurable (e.g. in terms of pystones). And using interrupts is quite more complex to implement. There isn't any hardware debounce AFAIK.