pybricks/support #718: [Bug] Pystone performance reduction

laurensvalk, 2022-08-31:
**Describe the bug**
The [pystone benchmark](https://github.com/pybricks/pybricks-micropython/tree/master/tests/pup/benchmark) has gone down by about 20% recently. It would be interesting to see what caused it. And maybe it could give us some ideas for other improvements.

Some recent changes worth looking into:
- The new I2C-IMU process that is running all the time.
- Download/SPI processes, although these should be idle except during boot/poweroff.
- Running the program without the explicit copy

**To reproduce**
Run the pystone benchmark. It used to be about 1000 pystones/second, now it is about 800.

We can go back in the git history to see if there was a major dropoff somewhere.

We can also check on other hubs to see if performance reduced in the same way or not.

laurensvalk, 2022-08-31:
Oldest to newer:

- https://github.com/pybricks/pybricks-micropython/commit/019d95c554706b831ca0af8d654b13dfee94fb56: 1012 pystone / second
- https://github.com/pybricks/pybricks-micropython/commit/c4979c44ea7107a6e49e9a9502695cc55a31263a: 768 pystone / seconds ----> New i2c imu driver seems to be a factor
- https://github.com/pybricks/pybricks-micropython/commit/da2154f8ec5e14bab4b58be0db8975ab3492cdde: 768 pystone / seconds 
- https://github.com/pybricks/pybricks-micropython/commit/f04ae8b6e87621cfea47878a3c89a21345be0979: 800 pystone / seconds ---> new download and run appears to be improvement

dlech, 2022-08-31:
For comparison, what is the change in performance if 6 motors are running vs no I/O attached?

laurensvalk, 2022-08-31:
I'll have to try that tomorrow.

Disabling the io ports altogether didn't seem to make a measurable difference.

laurensvalk, 2022-09-01:
It seems to cost about 25 to 30 pystone/second per motor running at a controlled 500 deg/s.

dlech, 2022-10-26:
Related IMU I2C experimentation: #753

laurensvalk, 2022-11-22:
Should we tackle this for V3.2 or defer?

dlech, 2022-11-22:
In the short term, we could add a delay between reads or not worry about it.

laurensvalk, 2022-11-22:
I'm OK not worrying about it :)

laurensvalk, 2022-12-20:
For Technic Hub, we're down to about 495, down from 700.

dlech, 2023-02-01:
As of https://github.com/pybricks/pybricks-micropython/commit/5942831fb45b906658cb4978dbac1c70d3daeda8

SPIKE Prime:

```
Pystone(1.2) time for 50000 passes = 47.675
This machine benchmarks at 1048.77 pystones/second
```

Technic:

```
Pystone(1.2) time for 50000 passes = 72.029
This machine benchmarks at 694.165 pystones/second
```

laurensvalk, 2023-06-08:
SPIKE Prime seems to have gone down to about `914.378 pystones/second`.

laurensvalk, 2025-10-07:
A few coffees later. It turns out the Boost Color and Distance sensor sends RGBI values at 1 kHz, which keeps the hub very busy.

The apparent culprit https://github.com/pybricks/pybricks-micropython/commit/8a52c950cfa29a03cc5f49ea24842a5ddce55359 just introduced default modes, including RGBI; it doesn't inherently slow things down.