pybricks/support #719: [Feature] Add dialog to thank our sponsors

laurensvalk, 2022-09-01:
It would be nice to thank our sponsors to show our appreciation for their contributions. We really do appreciate it!

This is very common practice with content creators on platforms such as Youtube. This is often done by listing the Patreon members at the end of the video.

Many of our users are not on GitHub, so this may also be a way to make the possibility of contributing a bit more visible.

Here's a suggestion on how we could do that:

[image]

Ideally we would read the data from an API provided by the respective service, so we can avoid storing personal data anywhere. This way contributors can also remove their listing by just unsubscribing from that particular service.

When Pybricks Code is offline, we could still show this but just omit the contributor listings.

laurensvalk, 2022-09-01:
For simplicity, we could start off without the contributor listing, and just make this a static dialog.

laurensvalk, 2022-09-27:
For inspiration: Inkscape now shows this on first boot:

[image]

laurensvalk, 2022-11-10:
Assign @laurensvalk, add some text.

dlech, 2022-11-18:
We now have a dialog that provides links to sponsorship pages, but I suppose we should leave this open if we still want to actually thank sponsors eventually.