pybricks/support #720: Can you uninstall The firmware that the app flash on to the brick?[Question]

luke-e-gibson-1, 2022-09-04:
**Question**

Can you uninstall The firmware that the app flashes onto the brick?
if I don't want the pybrick firmware any more?

luke-e-gibson-1, 2022-09-04:
I found a way to reflash with lego firmware here in this articale https://www.eurobricks.com/forum/index.php?/forums/topic/179604-how-did-you-fix-your-technic-hub-firmware-being-updated-every-time/

closed 2022-09-04 by luke-e-gibson-1

laurensvalk, 2022-09-04:
> **Question**
> 
> Can you uninstall The firmware that the app flashes onto the brick? if I don't want the pybrick firmware any more?

If anyone else is looking for this in the future, this is also covered on the installation instructions: https://pybricks.com/install/technic-boost-city/#restoring-the-original-firmware