pybricks/support #724: The firmware.metadata.py file contains missing or invalid entries. Fix it then try again.

DeusHyper, 2022-09-19:
Hey,

I am trying to install pybricks beta on an lego technic inventor hub and after setting it into DFU mode and setting up the driver correctly, it is shown as an "Lego Technic Large Hub" but I am sure it is an Inventor hub. Anyways, after I choose it to install the pybricks software on it, it gives me the error code from the title. I already reinstalled the original system with the pybricks restore tool which worked fine with the inventors hub software. Am I doing something wrong or is this an current issue?

OS: Windows 11
Hub: Inventors Hub from 2020

Kind regards,
Jonas Weber

dlech, 2022-09-19:
Can you give the exact steps you are doing, including details like version numbers, so we can try to reproduce the problem?

DeusHyper, 2022-09-20:
Yea sure I will do it later when I get home

DeusHyper, 2022-09-22:
Hey, forgot to update, sorry.
Did nothing but worked yesterday.
Thank you anyways!

closed 2022-09-22 by DeusHyper

estoykov, 2022-10-29:
Hi,

I have the same issue with Spike Essential Hub.

Windows 10 with USB drivers installed - as I already flashed the hub manually some time ago.
I even restored the latest original firmware and updated it with Spike application.

The last try was with (from both the web page and standalone PWA application): v3.2.0b4 (Pybricks Beta v2.0.0-beta.9)

I followed the steps from the web page, the hub is shown in the list;
[image]

But after "Connect", the error message appears:
"The firmware.metadata.py file contains missing or invalid entries. Fix it then try again."

Note that I tried with the previous beta versions with this web based firmware install, but they failed in the same way.

Cheers!
And keep the good work :)
Elijah

dlech, 2022-10-29:
I just tried it with v3.2.0b4 (Pybricks Beta v2.0.0-beta.9) and it is working for me. Are you using a custom firmware file?

estoykov, 2022-10-29:
Oops, it works but with the default hub name.
If you give a custom name to the hub ("Mini" in my case :) the mentioned error appears.

Cheers!

dlech, 2022-10-29:
Ok, wow, you found a hard-to-find bug! The combination of that exact firmware file and the name "Mini" made the checksum 0 and we made a rookie javascript mistake where 0 is treated as false.

estoykov, 2022-10-29:
Wow!

0 - like the green roulette zero - one of a kind? :)

And combined with JS code like "if (!checksum).." is just amazing ;)

I'm glad that helped you find this ;)
Cheers and keep the good work!

Elijah