pybricks/support #726: [Doc] Action buttons are too big

tiagorangel2011, 2022-09-25:
**Describe the issue**
The blue action buttons at the top are to big.

**Suggested improvement**
Make them smaller.

laurensvalk, 2022-09-25:
Could you add some information about your setup? What kind of device do you use? How big is the screen? Do you use an on-screen keyboard?

tiagorangel2011, 2022-09-25:
Hi! I'm currently using Windows 11 on a laptop, but I think this happens on
any computer.
You can find my browser details at https://whatismybrowser.com/w/5Y7ERX2

> Message ID: ***@***.***>
>

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
09/25/22,
11:24:26 AM

dlech, 2022-09-25:
Why do you think they are too big? Is it just aesthetics or is it causing a problem?

tiagorangel2011, 2022-09-26:
Ha, It's not causing any problems! 🤣 The only thing about it is that it
minimizes the coding area and it's a little difficult to concentrate. I
have marked this as low-priority

closed 2022-12-05 by dlech

dmusican, 2023-07-27:
Yes, this is fantastic, thank you! I'll open this up in a new issue.