pybricks/support #728: [Feature] Add animation or video for activating bootloader

laurensvalk, 2022-09-27:
**Is your feature request related to a problem? Please describe.**
The new interactive firmware installer has step-by-step instructions to prepare the hub for installing Pybricks.

Adding a video will help illustrate the steps and help confirm to the user that they've done it correctly.

**Describe the solution you'd like**
There are several ways we could do this. If we want to make the videos language-agnostic, we could do it like this.

We'd only need one video per hub this way, and we don't have to worry about text legibility in compression.

Thoughts?

[image]

laurensvalk, 2022-10-27:
https://user-images.githubusercontent.com/12326241/198326286-2e5ff499-36ad-4999-b354-276e6d689a7f.mp4

laurensvalk, 2022-10-27:
https://user-images.githubusercontent.com/12326241/198326553-29d5c0c1-7053-424c-87b4-a05fa3bf5d1c.mp4

laurensvalk, 2022-12-02:
It seems like vtt tracks allow putting in media references, which can then be loaded on cue:
https://www.w3.org/wiki/VTT_Concepts#Metadata_Tracks

dlech, 2022-12-02:
[vokoscreenNG-2022-12-02_14-50-37.webm](https://user-images.githubusercontent.com/963645/205384135-5eda1105-c102-4a62-8a59-08c715d701ae.webm)

Great find! This works brilliantly.

(We will need to slow down the pace of the videos to give time to read.)

laurensvalk, 2022-12-03:
Nice! That's the perfect balance between having the text always visible vs. syncing them like subtitles.

I stumbled on this completely coincidentally. I was trying to fix some CSS in our Sphinx compatibility animation. I got completely stuck but at least it wasn't all for naught then :smile:

laurensvalk, 2022-12-03:
https://user-images.githubusercontent.com/12326241/205461234-9294ff99-5132-45af-a587-58602dd6028f.mp4

laurensvalk, 2022-12-03:
960x540 but lesser quality, 1.6 MB, so about 2 MB/min.

https://user-images.githubusercontent.com/12326241/205461622-c0c4de95-a120-480b-81dd-88eacbd62c69.mp4

laurensvalk, 2022-12-03:
HD is probably overkill. The second one is still nice. The third one could be fine if we want to save on download size.

dlech, 2022-12-05:
Can we just make the videos 600 pixels wide?

laurensvalk, 2022-12-06:
Sure, if that works best for the app.

laurensvalk, 2022-12-08:
https://user-images.githubusercontent.com/12326241/206546870-1a254eee-3e28-4b38-9345-0cb17baea6b1.mp4

laurensvalk, 2022-12-09:
https://user-images.githubusercontent.com/12326241/206737878-fe00a7ee-3cde-430d-8297-f849be45a6de.mp4

https://user-images.githubusercontent.com/12326241/206737881-f2e758b1-9619-4f39-8dd3-fb19bf5b683d.mp4

https://user-images.githubusercontent.com/12326241/206737885-a50869c0-cf4f-41d7-88c6-1241e0f8cccb.mp4

laurensvalk, 2022-12-09:
This should be all the clips we need to begin with.

I have hidden all comments with test videos that should not be used.

laurensvalk, 2022-12-09:
Maybe we can put these assets in a folder and add a [CC BY-NC-SA](https://creativecommons.org/licenses/by-nc-sa/4.0/) license text in that folder?

closed 2022-12-09 by dlech