pybricks/support #729: [Bug] DriveBase acting weird with high values?

Vinz1911, 2022-09-27:
**Describe the bug**
I have a Spike prime with two large motors equipped. The DriveBase is starting acting weird after I set a high speed value or a high turn rate or both at the same time. For example: if I just add a speed value of 500 the robot accelerate very fast for a second and then drives slow ?!. If I add a high turn rate it's the same behavior, it starts turning fast for a second and then it drives/turns slow and sometimes it stops and starts turning in the wrong/opposite direction ?!. If I set a speed value of 300 and a turn rate of 45 (or values below) then everything seems to working correct.

As additional Question: If I set an acceleration value in the settings this does not affect the `drive()` method. it would be nice if I can also set the acceleration value as a setting for the `drive()` method. the default value is a bit slow for me.

**To reproduce**
This little Code snippet is enough to reproduce the behavior, sometimes it needs to restart a few times to trigger it.
```python
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Button, Color, Direction, Port, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

hub = PrimeHub()
left_motor = Motor(Port.A)
right_motor = Motor(Port.B, Direction.COUNTERCLOCKWISE)

base = DriveBase(left_motor, right_motor, 55, 125)

# high drive speed
base.drive(500, 45)

# or high turn rate or both
base.drive(500, 70)

while True: pass
```
**Expected behavior**
Drive as expected with high speed and correct turn rate

laurensvalk, 2022-09-27:
Thanks for reporting. Can you add the firmware version used for this experiment?

```python
from pybricks import version
print(version)
```

Vinz1911, 2022-09-27:
Yes sure:

```python
('primehub', '3.2.0b3', 'v3.2.0b3-67-gcaffa8a6 on 2022-09-14')
```

laurensvalk, 2022-09-27:
Could this be the same as #664 ?

I thought I had fixed it but apparently not.

Vinz1911, 2022-09-27:
So Problem also does NOT occur with the normal Spike Motors, I have it only with the Large ones.

It seems, this can be the Problem :D

laurensvalk, 2022-09-29:
I can reproduce this, but it is separate from #664.

In #664, there is strictly an issue about the pre-computed trajectory. The reference position jumps, but the PID controller still does as it's told: it goes to the bad trajectory.

But in this issue we have e.g:

```
motor.run(1000)
```
> then you also can reproduce it, it starts accelerating fast for a second and then keeps running on a much slower speed

So there is something wrong with the PID controller here.

laurensvalk, 2022-09-29:
Possibly the same cause. Run until stalled no longer works:

```python
from pybricks.pupdevices import Motor
from pybricks.parameters import Port

steer = Motor(Port.A)
left_end = steer.run_until_stalled(200)
```

laurensvalk, 2022-10-04:
[image]

> So there is something wrong with the PID controller here.

The torque output is overflowing.

laurensvalk, 2022-10-04:
It looks much better after the fix :smile: 

[image]

laurensvalk, 2022-10-04:
[This build](https://github.com/pybricks/pybricks-micropython/actions/runs/3182851071) fixes the stall issue shown above.

It may or may not also fix the issue you saw.

Vinz1911, 2022-10-04:
This is absolutely Insane!!! 

Your fix work very well, it's like driving the robot on rails!
Also on high RPM no problems at all :)

laurensvalk, 2022-10-04:
Sounds like you’ve discovered the full potential of the recent drive base updates we’ve been working on 😄

laurensvalk, 2022-10-04:
Out of curiosity, since you’re the probably one of the first to try the new drive base, what is it that you like in particular? The new features or mainly the precision and performance?

laurensvalk, 2022-10-18:
Run until stalled is now fixed as well, so I think we can close this.

closed 2022-10-18 by laurensvalk