pybricks/support #730: [Question] Robot Inventor died/bricked after DFU

fuzzycow, 2022-09-28:
**Question**
Robot Inventor Hub appears to have died/bricked immediately after running the DFU (which completed successfully).
No lights on the HUB. Attempted charging for a couple of days - no improvement.

Has anyone experienced this and may offer a suggestion ?

laurensvalk, 2022-09-28:
Remove the battery and plug it back in. Then you can run the DFU update again.

Which page did you use for the DFU install?

laurensvalk, 2022-09-28:
Can you get it into DFU mode again?

- Remove the battery, and plug it back in. This is normally not necessary, but this is just so we know the hub is really off. 
- Press and hold the Bluetooth button.
- While you hold it, plug in the USB cable. Make sure the other end is plugged into your computer.
- keep holding the Bluetooth button. The light should turn pink/purple.
- keep holding it. After a few seconds, the light should start blinking green/red/blue. 

Do the lights behave in this way or are you seeing something else?

fuzzycow, 2022-09-28:
The process you described worked, and the HUB is operational again. I managed to DFU with: https://dfu.pybricks.com/firmware-backup-v1.4.01.0000-594ce3d.bin.
Thank you so much!

Originally the issue occurred after: DFU via https://dfu.pybricks.com/, firmware: https://dfu.pybricks.com/essential-v1.0.00.0070-51a2ff4.bin (in retrospect probably not the right firmware for Robot Inventor).

laurensvalk, 2022-09-28:
Glad to hear it!

And thanks for taking the time to share details about your first approach. That helps us rule out technical issues :)

We're planning to integrate the restore process in the pybricks app in such a way that the correct file is automatically selected.

closed 2022-09-29 by laurensvalk