pybricks/support #733: [Feature] Use apt install for gcc-arm-none-eabi

laurensvalk, 2022-10-04:
**Is your feature request related to a problem? Please describe.**
This should help avoid the CI get stuck sometimes:

[image]

laurensvalk, 2022-10-06:
Related - I've been having to do this to get intellisense to work, but I'm not sure this is all needed.

```diff
diff --git a/.vscode/c_cpp_properties.json b/.vscode/c_cpp_properties.json
index 4b4d4ced..8e696aa0 100644
--- a/.vscode/c_cpp_properties.json
+++ b/.vscode/c_cpp_properties.json
@@ -1,14 +1,10 @@ed in makefiles
-// * compiler options should match what is selected in the makefiles
 {
     "configurations": [
         {
             "name": "primehub",
             "includePath": [
+                "/usr/lib/gcc/arm-none-eabi/10.3.1/include",
+                "/usr/lib/gcc/arm-none-eabi/10.3.1/include-fixed",
                 "${workspaceFolder}/micropython/lib/cmsis/inc",
                 "${workspaceFolder}/micropython/lib/stm32lib/CMSIS/STM32F4xx/Include",
                 "${workspaceFolder}/micropython/lib/stm32lib/STM32F4xx_HAL_Driver/Inc",
@@ -37,6 +33,7 @@
                 "STM32_HAL_H=<stm32f4xx_hal.h>",
                 "USE_FULL_LL_DRIVER"
             ],
+            "compilerPath": "/usr/bin/arm-none-eabi-gcc",
             "compilerArgs": [
                 "-Wall",
                 "-Werror",
@@ -53,7 +50,7 @@
                 "-fsingle-precision-constant"
             ],
             "cStandard": "c11",
-            "intelliSenseMode": "gcc-arm"
+            "intelliSenseMode": "linux-gcc-arm"
         },
```

dlech, 2022-10-06:
We should not include OS paths in this file since not everyone has the same OS. And since our targets are bare metal, the intellesense mode should not be linux. Is it enough to just set `"compilerPath": "arm-none-eabi-gcc"`?

dlech, 2022-10-06:
Maybe check the C/C++ logs in the output pane? Intellesense has been working for me with the current config on Ubuntu 22.04 using the system gcc-arm-none-eabi. Or maybe a reboot solves the issue?

laurensvalk, 2022-10-10:
From what I could find, all docs match the pop-up above to say that the full path should be used.

laurensvalk, 2022-10-10:
It seems that I can set
```
"C_Cpp.default.compilerPath": "/usr/bin/arm-none-eabi-gcc",
```
In `.vscode/settings.json`, which isn't committed. This works both in project settings and user settings.

Maybe you've got something like that (or a different) setting in your settings too somewhere?

This workaround is fine for me though.

dlech, 2022-10-10:
No, I don't have any settings like that configured. But it that fixes it for you, that good.

closed 2022-10-10 by dlech