pybricks/support #734: [Bug] example motor_until_stalled_center.py does not center the steering (it is not in the program)

BertLindeman, 2022-10-08:
In beta.pybricks.com looked at the doc (v3.2) for an example to center the steering of a model.
Surprised that the `examples/pup/motor/motor_until_stalled_center.py` does not do the centering.

The examples `motor_until_stalled_center.py` and `motor_until_stalled.py` are **the same**.

In code.pybricks.com these examples are also the same in the doc (v3.1).

Maybe the original example (I thought is WAS there) got lost?

laurensvalk, 2022-10-08:
This is related to #729, but there is also an additional bug that I'm in the process of fixing.

As a workaround, try the (latest) firmware given in #729, and *additionally* add a `stop()` call in between successive `run_until_stalled` calls.

Thanks for reporting!

laurensvalk, 2022-10-10:
The docs are fixed but the firmware is not, so re-open.

laurensvalk, 2022-10-18:
> The docs are fixed but the firmware is not, so re-open.

This is fixed --- the example now runs on the latest firmware again.

closed 2022-10-18 by laurensvalk