pybricks/support #735: [Bug] After update with CI firmware build 2071 hubs fail to run programs with "timeout waiting for checksum"

BertLindeman, 2022-10-12:
**Describe the bug**
"Run program" fails with "timeout waiting for checksum" with CI-firmware in build 2071
Tested MoveHub, CityHub and RobotInventorHub.

**To reproduce**
Steps to reproduce the behavior (Win11 at 22H2; Google Chrome Version 106.0.5249.103 (Official Build) (64-bit)):
1. Go to beta.pybricks.com
2. Click on "Connect to Bluetooth" to a hub with this firmware
3. Pair the hub
4. Click on F5 to run the program (see below)
5. In the browser open Developer tools (CTRL-SHIFT-I)
6. see the timeout 

**Expected behavior**
The python program to run
The simplest program:
```python
print("X")
```

Same result on Linux Mint Vanessa (22.04) and Google Chrome 106.

**Extra info**
Firmware at ```('primehub', '3.2.0b3', 'v3.2.0b3-55-gc126dd6b on 2022-08-26')``` works OK.
Both on wodows and linux mint.

**Screenshots**
[image]

laurensvalk, 2022-10-12:
Thanks for reporting!

We recently changed the download-and-run protocol to:
a) speed it up
b) add flexibility to add more advanced features later

Pybricks Code hasn't been updated yet, so you may wish to stick to an older firmware for the moment.

laurensvalk, 2022-10-12:
If you don't mind using the latest cutting edge (master branch, not released), then yes, you can use Pybricksdev :smile:

Is there any particular feature you want to try?

BertLindeman, 2022-10-12:
> Is there any particular feature you want to try?

Not really, Laurens. But I have a program using a technic hub and the remote to drive a model.
And THAT program does strange things like stopping without notice, or stating that the stop button has been pressed.
Mostly on lines that do motor manipulations, like as example a run_until_stalled.
At firmware level of beta.pybricks.com

just trying to get a finger behind it. So strange.

laurensvalk, 2022-10-12:
> stopping without notice, or stating that the stop button has been pressed.

Like https://github.com/pybricks/support/issues/716, or different (later on in a program?)

Is it the same behavior/error that occurs when you unplug a motor while it is running?

To make things easier, you can try this version. It has everything except the download-and-run changes. This means you can still use it with the current version of Pybricks Beta:
https://github.com/pybricks/pybricks-micropython/actions/runs/3219255635

BertLindeman, 2022-10-12:
After the program starts and remote made connection there are a number of run_until_stalled and setting motor angles to calibrate the things.

No settings for buttons, nor any button press is done.

Will try the firmware you point to.

Thank you.

BertLindeman, 2022-10-12:
> To make things easier, you can try this version. It has everything except the download-and-run changes. This means you can still use it with the current version of Pybricks Beta:
> https://github.com/pybricks/pybricks-micropython/actions/runs/3219255635

This firmware works OK for me, no more strange things seen (yet 😉 ) 
Firmware indicated as build 2063.
Thank you.

closed 2022-10-12 by BertLindeman