pybricks/support #737: [Bug] Unexpected error when connecting too fast.

laurensvalk, 2022-10-18:
**Original issue**
When connecting to a hub after turning it on, there is often a Bluetooth error. Trying again fixes it.

This has happened for as long as I remember, but reproducing it has been an issue.

**Reproducible variant**
When quickly connecting to a hub after disconnecting, there is often a Bluetooth error. Trying again fixes it.

This seems quite similar.

**To reproduce**

The following is done with an Inventor Hub with the latest firmware, but I have previously seen this happen with all hubs.

[Screencast from 18-10-22 09:38:10.webm](https://user-images.githubusercontent.com/12326241/196367650-f63c75c0-c0b6-44e4-b3af-552d6dcad81f.webm)

Is it possible to wait a while for `getPrimaryService` to get ready?

dlech, 2022-10-18:
What operating system are you using? We have quite a few reports already for this error on Windows.

We've already added a 1 second delay in https://github.com/pybricks/pybricks-code/commit/a86967573c35fc9b5e2bc1786ab425db9e603207 to try to work around this.

Can you include Wireshark logs, D-Bus logs, etc?

laurensvalk, 2022-10-18:
Ubuntu 22.04.

I might be able to get logs on Thursday or Friday.

dlech, 2022-10-21:
Does https://github.com/pybricks/pybricks-micropython/commit/1815b8e6b65ca46ae20a9b41dc9b98a4b22735d4 make a difference?

laurensvalk, 2022-10-21:
> Does https://github.com/pybricks/pybricks-micropython/commit/1815b8e6b65ca46ae20a9b41dc9b98a4b22735d4 make a difference?

Doesn't seem like it.

laurensvalk, 2022-10-21:
> We've already added a 1 second delay in https://github.com/pybricks/pybricks-code/commit/a86967573c35fc9b5e2bc1786ab425db9e603207 to try to work around this.

Interestingly, even a 10 second delay *here* doesn't help.

By contrast, waiting 10 seconds *before* clicking the button always works.

dlech, 2022-10-26:
Since we have made an improvement, should we call this good enough and close the issue?

closed 2022-10-27 by laurensvalk