pybricks/support #738: [Bug] print() while the hub is disconnected can briefly hang the hub.

laurensvalk, 2022-10-18:
**Describe the bug**
Doing a lot of `print` statements while the hub is disconnected can briefly hang the hub.

The data is supposed to be ignored when the buffer is full but apparently that is not quite happening as intended.

To reproduce, run this on a SPIKE "drivebase", or just attach two motors and let them spin freely. Run it **without** Pybricks Code connected. When it starts printing at the end, you can consistently see the 3x3 matrix animation get stuck briefly.

Occasionally (not often), it also pushes the motors out of sync (https://github.com/pybricks/support/issues/679), which is how I first noticed this.

```python
from pybricks.pupdevices import Motor
from pybricks.tools import wait
from pybricks.parameters import Port, Direction
from pybricks.robotics import DriveBase
from pybricks import version

print(version)

# Initialize default "Driving Base" with medium motors and wheels.
left_motor = Motor(Port.C, Direction.COUNTERCLOCKWISE)
right_motor = Motor(Port.D)
drive_base = DriveBase(left_motor, right_motor, wheel_diameter=56, axle_track=112)

# Allocate logs for motors and controller signals.
DURATION = 6000
left_motor.log.start(DURATION)
right_motor.log.start(DURATION)
drive_base.distance_control.log.start(DURATION)
drive_base.heading_control.log.start(DURATION)

# Drive straight forward and back again.
drive_base.straight(500)
drive_base.straight(-500)

# Wait so we can also log hold capability, then turn off the motor completely.
wait(100)
drive_base.stop()

# Transfer data logs.
print("Transferring data...")
left_motor.log.save("servo_left.txt")
right_motor.log.save("servo_right.txt")
drive_base.distance_control.log.save("control_distance.txt")
drive_base.heading_control.log.save("control_heading.txt")
print("Done")
```

dlech, 2022-10-18:
Are we sure the problem is with printing and not `log.save`? When there is no connection, it will loop through all log lines without ever calling the contiki event loop (normally, printing will block when the buffer is full and call `MICROPY_EVENT_POLL_HOOK`).

We already have:

```c
        // Writing data can take a while, so give MicroPython some time too
        mp_handle_pending(true);
```

in this loop. Maybe we need to change it to `MICROPY_EVENT_POLL_HOOK`?

laurensvalk, 2022-10-18:
Good call, will try it out.

laurensvalk, 2022-10-18:
That seems to work. The connected case does seem to run a lot slower that way. I'd have to dig into why that happens but I can live with that for now.

Patch incoming.

dlech, 2022-10-18:
`MICROPY_EVENT_POLL_HOOK` waits for an interrupt before continuing. So we could consider calling:

```c
mp_handle_pending(true);
MICROPY_VM_HOOK_LOOP
```
 
instead, which won't wait.

closed 2022-11-11 by laurensvalk