pybricks/support #740: [Bug] GATT operation failed for unknown reasons. with Spike Prime Hub

davidwu226, 2022-10-19:
**Describe the bug**
After upgrading pybricks-micropython to b83536e69693e64e312343ec6b9a79c513fdc24f, when I try to use bluetooth to connect on code, I get an error GATT operation failed for unknown reason. The previous commit at ed2e5849f52211f0e3e98d269e861c4fb08752c2 seems to work fine.

**To reproduce**
Steps to reproduce the behavior:
1. Check out pybricks-micropython at b83536e69693e64e312343ec6b9a79c513fdc24f
2. Deploy to Spike Prime Hub.
3. Go to beta.pybricks.com and connect via Bluetooth.
4. See error.

**Expected behavior**
The editor should have connected to the hub via bluetooth.

**Screenshots**
[image]

dlech, 2022-10-20:
What operating system are you using?

davidwu226, 2022-10-20:
I'm on Mac OSX 12.4 (M1-based) with Chrome 106.0.5249.119.

Here's from the copy error message:

```
GATT operation failed for unknown reason.
```

laurensvalk, 2022-10-20:
> Check out pybricks-micropython at https://github.com/pybricks/pybricks-micropython/commit/b83536e69693e64e312343ec6b9a79c513fdc24f

This is in the middle of big revisions to the Bluetooth code.

Before digging into this, you may want to try:
- https://github.com/pybricks/pybricks-micropython/commit/0aeb8bf8fcbda869e462642ee3fb833d3e41be94: This is just before we started revising bluetooth extensively. This will still work with beta.pybricks.com as it is today.
- https://github.com/pybricks/pybricks-micropython/commit/7d7434ff5404e94183ede83ea0f1cbddfa321f28: This is the latest version on the master branch. This completes the Bluetooth revisions but it is not yet compatible with beta.pybricks.com. This should come in a week or two when we release the next beta version.

davidwu226, 2022-10-20:
Thanks @laurensvalk for responding so quickly! As I mentioned, the previous commit (ed2e584) worked without problems.

FWIW, I also tried pybricks/pybricks-micropython@b83536e against pybricks/pybricks-code@81d52db and it did not work either.

laurensvalk, 2022-10-20:
> As I mentioned, the previous commit (ed2e584) worked without problems.

Right, anything before those Bluetooth changes is supposed to work with https://beta.pybricks.com/ up to today.

> And yes, I did try the latest version of the master branch (which also does not work) 

No matching Pybricks Code version was released (or even pushed) at the time, I think.

But since you are able to run Pybricks Code locally, it would be great if you want to try the latest version of the master branches of both, together.

To be specific, e.g.:

https://github.com/pybricks/pybricks-micropython/commit/57efb61b3f33e6f05abb03617b49d5963a825cdf

with

https://github.com/pybricks/pybricks-code/commit/7e726fffafc21e08c37468d00a4f2378f0957f91

Thanks!

I've been running those all day today. Would be great to know if the issue persists in that case.

davidwu226, 2022-10-20:
Just tried pybricks/pybricks-micropython@57efb61 with pybricks/pybricks-code@7e726ff still not working:

```
No Characteristics matching UUID c5f50003-8280-46da-89f4-6d8051e4aeef found in Service with UUID c5f50001-8280-46da-89f4-6d8051e4aeef.
```

dlech, 2022-10-20:
Can you do a Bluetooth packet capture as described at https://bleak.readthedocs.io/en/latest/troubleshooting.html#macos while trying to connect?

We changed the layout of the GATT attributes on the device and I have a hunch that macOS may be serving up a cached attribute database instead of reading it from the hub.

davidwu226, 2022-10-20:
Sure, here's the capture from PacketLogger:

[btcap.zip](https://github.com/pybricks/support/files/9833007/btcap.zip)

davidwu226, 2022-10-20:
Bummer... spoke too soon. Now the Bluetooth connects but the hub seems to crash and reset when I try to open up the interactive Python session.

dlech, 2022-10-20:
Thanks! I'm working on a fix to use random Bluetooth addresses to avoid the caching issues.

The REPL issue is also being worked on in #741.

laurensvalk, 2022-10-20:
> Bummer... spoke too soon. Now the Bluetooth connects but the hub seems to crash and reset when I try to open up the interactive Python session.

REPL shouldn't currently crash, so feel free to open an issue if that persists.

laurensvalk, 2022-10-20:
Thanks for testing on the cutting edge. Hopefully it doesn't spoil your first impression of Pybricks :sweat_smile: .

dlech, 2022-10-20:
> Bummer... spoke too soon. Now the Bluetooth connects but the hub seems to crash and reset when I try to open up the interactive Python session.

See https://github.com/pybricks/support/issues/744#issuecomment-1286030639 for workaround.

dlech, 2022-11-26:
It seems like this issue is resolved. Let us know if it isn't.

closed 2022-11-26 by dlech

mrinalraghupathi, 2023-01-30:
I get the same error (randomly but not always) while running the following code on the robot inventor hub.

Mac OS: Ventura 13.2 (running Intel i5)
Chrome: Version 109.0.5414.119 (Official Build) (x86_64)

## Code being run
```
from pybricks.hubs import InventorHub
from pybricks.pupdevices import Motor, ColorSensor, UltrasonicSensor
from pybricks.parameters import Button, Color, Direction, Port, Side, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

hub = InventorHub()

hub.light.on(color=Color.RED)

wait(1000)
```

## Error message
```
GATT operation failed for unknown reason.
```

mrinalraghupathi, 2023-01-30:
I was making a screencast to show folks how to use pybricks at the same time. It seems the error only happens when the screencast software is also running (OBS studio).