pybricks/support #741: [Bug] REPL auto-import does not work.

laurensvalk, 2022-10-20:
When starting the REPL, everything in the `pybricks` package is supposed to be pre-imported.

This no longer works because:

```c
    for (size_t i = 0; i < MP_ARRAY_SIZE(pybricks_globals_table); i++) {
        mp_rom_obj_t module = pybricks_globals_table[i].value;
        if (mp_obj_is_type(module, &mp_type_module)) {
            // Import everything from the module.
            mp_import_all((mp_obj_t)module);
        }
    }
```

and the modules are no longer included in `pybricks_globals_table`.

laurensvalk, 2022-10-20:
> While we are doing this, we could consider importing everything, including those from MicroPython modules. I'll have a look at the RAM cost

On Prime Hub, when importing everything:

```
stack: 604 out of 36976
GC: total: 259008, used: 2272, free: 256736
 No. of 1-blocks: 14, 2-blocks: 3, max blk sz: 112, max free sz: 15915
```

When importing only pybricks modules (+36 bytes build size for the check):

```
stack: 604 out of 36976
GC: total: 259008, used: 1008, free: 258000
 No. of 1-blocks: 20, 2-blocks: 4, max blk sz: 19, max free sz: 16116
```

So, about 1.2K extra heap used for importing everything from the (currently included) MicroPython modules.

Now check on Move Hub:

dlech, 2022-10-20:
Are we sure that there are and never will be name conflicts for importing * from *? It seems like it would be safer to import * from pybricks modules but import module for everything else.

laurensvalk, 2022-10-20:
OK, I'll restore importing Pybricks modules only.

dlech, 2022-10-20:
I like the idea of importing the other modules too. Maybe we could import * if module starts with pybricks else import module?

laurensvalk, 2022-10-20:
Sure. Will do that tomorrow.

laurensvalk, 2022-10-20:
`mp_import_name` and several other related calls just fetch the module but don't register it as loaded. Simply doing it like https://github.com/pybricks/pybricks-micropython/commit/7b57a7a18d75e2c80ec32e6400fef19b0d5ce3f3 works though.

dlech, 2022-10-21:
can this be closed now?

laurensvalk, 2022-10-21:
Yes.

closed 2022-10-21 by laurensvalk