pybricks/support #744: starting REPL crashes Prime hub

dlech, 2022-10-20:
> Bummer... spoke too soon. Now the Bluetooth connects but the hub seems to crash and reset when I try to open up the interactive Python session.

> The REPL issue is also being worked on in https://github.com/pybricks/support/issues/741.

REPL shouldn't currently crash, so feel free to open an issue if that persists.

_Originally posted by @laurensvalk in https://github.com/pybricks/support/issues/740#issuecomment-1286020333_

dlech, 2022-10-20:
I was able to reproduce this as well.

- Flash SPIKE Prime hub with https://github.com/pybricks/pybricks-micropython/commit/57efb61b3f33e6f05abb03617b49d5963a825cdf
- Connect hub to Pybricks code from https://github.com/pybricks/pybricks-code/commit/7e726fffafc21e08c37468d00a4f2378f0957f91
- Press the REPL button
- Hub crashes and resets before any output printed to Pybricks Code

laurensvalk, 2022-10-20:
Possibly relevant: It's also worth doing some checks with the data area erased (using the official firmware).

dlech, 2022-10-20:
Confirmed that running a user program with the :arrow_forward: button then trying the REPL again fixes the problem.

laurensvalk, 2022-10-21:
It would be good to review this carefully, also in situations where power is lost while loading a large program, after the write flag is already set due to another program having just run.

Maybe it is all already taken care of; I just haven't double-checked this after the recent revisions to download-and-run.

laurensvalk, 2022-11-22:
We also need to check that you can't accidentally erase half of your user program when entering bootloader mode.

This doesn't seem to be happening on Move Hub. Still need to try City Hub, Technic Hub.

laurensvalk, 2024-03-11:
> This doesn't seem to be happening on Move Hub. Still need to try City Hub, Technic Hub.

Entering bootloader does not erase flash, so there is nothing left to be done here. Closing.

closed 2024-03-11 by laurensvalk