pybricks/support #745: [Bug] Cannot import last-used main module on REPL

laurensvalk, 2022-10-20:
This is somewhat of a minor issue, but while we are looking at the REPL:

Since we started using the `__main__` name to search for the entry point (as opposed to using the first or last script), the last-used main script cannot be imported by its original name. It also can't be imported as `__main__` because that is now the REPL itself.

The use case isn't entirely far-fetched though. You could think of using the REPL to prepare test conditions, and then try to import your script to see how it behaves.

dlech, 2022-10-21:
Note for when we work on this: in pybricksdev, we would need to replace `finder.run_script(path)` to `finder.load_file(path)` to get the actual module name instead of `__main__`.

laurensvalk, 2024-10-02:
I suppose the behavior that we have now is technically correct.

Since you can enter the REPL from a main script with `raise KeyboardInterrupt`, the use case described here becomes a bit of an edge case anyway.

So I think we can close this.

closed 2024-10-02 by laurensvalk