pybricks/support #746: multi-file programs fail to run on move hub

dlech, 2022-10-21:
**Describe the bug**
Trying to run a program with multiple modules causes the move hub to reboot due to the watchdog timer.

**To reproduce**
Steps to reproduce the behavior:

`a.py`

```python
class A:
    m1 = 1
    m2 = 2
    m3 = 3

print('ran a as', __name__)
```

`b.py`

```python
from a import A

print('ran b as', __name__)
```

Running `a.py` works but running `b.py` causes the hub to reboot.

**Expected behavior**
`b.py` should work as on other hubs.

**Additional Context**

I checked all the way back to https://github.com/pybricks/pybricks-micropython/commit/542a47004df08a84eb2b25e74839151b7af0d00a and it is still broken there, so more recent changes didn't break it. Can't check back farther than that since that was when the new download and run protocol was introduced.

dlech, 2022-10-21:
Seems to affect the city hub as well, so maybe something to do with the F0 processors?

closed 2022-10-21 by dlech

laurensvalk, 2022-10-21:
Maybe that padding we previously had wasn’t all for naught then :)

This solution is probably fine too. Apparently MicroPython doesn’t mind being unaligned.