pybricks/support #750: 'Download and run' progress animation starts (much) later on Pybricks Beta v2.0.0-beta.6 firmware

BertLindeman, 2022-10-24:
### Discussed in https://github.com/orgs/pybricks/discussions/749

<div type='discussions-op-text'>

<sup>Originally posted by **BertLindeman** October 24, 2022</sup>
Not sure if this **is** a defect or feature or not, so started a discussion about this.

(Very nice that beta.pybricks.com now handles newer firmware again. Happy!)

The `download and run` button in ```v3.2.0b4 (Pybricks Beta v2.0.0-beta.6)```  takes longer to start the circle progress animation..
I often wonder if I **really** clicked the button.

As an example: it takes about 4 seconds after "click" for the progress circle comes up.
This is a program of just over 400 lines.
Assume this has to do with loading / including other sources.

**Remarkable**

1. Connected to Robot Inventor hub at beta6 firmware ```('primehub', '3.2.0b4', 'v3.2.0b4 on 2022-10-21')``` the button reacts very slow.

2. Connected to Spike Prime at CI firmware 2016  ```('primehub', '3.2.0b3', 'v3.2.0b3-55-gc126dd6b on 2022-08-26')``` this behaviour is "normal" and feels snappy. </div>

My testcase for this issue (in no way a good program):
```python

# program structure:
# init program
#   set constants
#   connect remote
#   calibrate gearbox
#   calibrate steering
#   set mode drive
# run in a loop
#   check and process buttons on the remote
#   wait a bit

# /BL 2022-10-22 try to add a function to determine
#                the correct gearbox motor angles

from pybricks.hubs import ThisHub
from pybricks.parameters import Port, Button, Stop, Color, Direction
from pybricks.pupdevices import Motor, Remote
from pybricks.tools import wait, StopWatch
from pybricks import version

print(version)

# set constants
DEBUG = True  # or False
GEARBOX_ROTATE_SPEED = 350  # gearbox motor rotate speed
STEER_ROTATE_SPEED = 200    # steering motor rotate speed
DRIVE_WITH_SPEED = 1000     # Speed in drive mode

# Used for calibration of gearbox motor:
TEST_SPEED = 350  # was 150
TEST_TIME = 3000

# mode names
MODEDICT = {0: "driveSlow", 1: "driveFast", 2: "height", 3: "winch"}
# motor angle per mode
# gearbox angle zero is for the winch

MODEANGLES = {0: 100, 1: 180, 2: 0, 3: 0}
# remote and hub LED color to show the current mode
MODECOLORS = {0: Color.GREEN, 1: Color.YELLOW, 2: Color.MAGENTA, 3: Color.GRAY}

# Motordirection for checking the stall of the gearbox, speed for the drive motors
motor_direction = 1  # Or -1 to turn left or right (flip-flop)

mode = 0    # 0 - drive mode
[... 333 lines cut ...]
        # Wait just a bit
        wait(50)

except Exception as e:
    print('Got exception:', e)
    # global remote
    # global hub
    #
    # stop the motors
    motorA.stop()
    motorB.stop()
    steer.stop()
    gearbox.stop()
    # drop connection to the remote
    # /BL Rationale:
    #     Sometimes the program on the hub runs on after "stop" on beta.pybricks.comp
    #     and only stops if the battery of the remote is pulled out.
    del remote
    del hub
    print("End of exception processing.")
finally:
    print("Leaving the program in 'finally'.")
    SystemExit()

```

laurensvalk, 2022-10-24:
The simple recursive search we originally had in pybricksdev might be an option.

It had some shortcomings for certain special cases that it didn’t cover yet, but I suppose we could cover those by handling them properly?

BertLindeman, 2022-10-24:
> Here is a perf snapshot. It blocks for 2 seconds with the program above on my development machine. 

Looks like my machine is a tad slower 😁 
[image]

closed 2022-10-24 by dlech

BertLindeman, 2022-10-24:
Fast fix, David.
At Pybricks Beta v2.0.0-beta.7 performance now shows:
[image]