pybricks/support #752: [Bug] Spike Prime medium motors with DriveBase appear to be out of sync in motion at low speeds

davidwu226, 2022-10-25:
**Describe the bug**
When I use DriveBase.drive(10, 0) (i.e., low velocity) with a Spike Prime medium motor, I can see that the left and right motors appear to be jerky (start-stop-start motion) and they also do not appear to turn at the same time (the start-stop-start motion are out of sync from the left motor to the right motor).

**To reproduce**
Steps to reproduce the behavior:

Use a Spike Prime medium motors with DriveBase.

``
lm = Motor(Port.A, Direction.COUNTERCLOCKWISE)
rm = Motor(Port.B, Direction.CLOCKWISE)
db = DriveBase(lm, rm, 57, 15 * 8)
db.drive(10, 0)
wait(10000)
``

Observe the wheels turning in a jerky manner and out of sync.

**Expected behavior**

Smooth slow motor turns.

dlech, 2022-10-25:
Which firmware version are you using?

davidwu226, 2022-10-25:
Pretty recent: pybricks/pybricks-micropython@7fb893190

davidwu226, 2022-10-25:
I also just tested drive(10, 0) with the Spike Prime large motors for the left and right motors and saw the same start-stop-start and out of sync behavior as I did with the medium motors.

laurensvalk, 2022-10-25:
```python
db.drive(10, 0)
```

10 mm/s with a wheel diameter of 57 mm is about 20 degrees per second. That is only about 2% of the rated speed (about 1000 deg/s).

Due to how they are built, these motors don't do great below 100 deg/s, and pretty much not at all below 50 deg/s.

If you want to drive at 10 mm/s, it might be better to gear the motors down.

laurensvalk, 2022-12-07:
This question has been answered, so we can close it.

closed 2022-12-07 by laurensvalk