pybricks/support #753: [Bug] Spike Prime Hub i2c seems to hang

davidwu226, 2022-10-25:
**Describe the bug**
I've been working on adding gyro angle support to the code base (with filtering to address gyro drift) on the Spike Prime Hub. I noticed that the i2c reads of one byte would hang if you do back to back i2c reads too quickly. The hub still appears to work but there doesn't seem to be a response back from the read. To work around this issue, I changed the reads to two byte and it seems to work.

**To reproduce**
Steps to reproduce the behavior:

My code would do a read of the gy read flag:

```
static PT_THREAD(until_gy_rdy(struct pt *pt)) {
    static struct pt child;
    static uint8_t rdy;

    pbdrv_imu_dev_t *imu_dev = &global_imu_dev;

    PT_BEGIN(pt);
    do {
        PT_SPAWN(pt, &child, lsm6ds3tr_c_gy_flag_data_ready_get(&child,  &imu_dev->ctx, &rdy));
    } while (!rdy);

    PT_END(pt);
}

```

Calling it in a repeated loop until ready:

```
   do {
        PT_SPAWN(pt, &child, lsm6ds3tr_c_gy_flag_data_ready_get(&child,  &imu_dev->ctx, &rdy));
    } while (!rdy);
```

This would work for a little but and eventually would never return.

The workaround fix I made to get it working for now:

```
 PT_THREAD(lsm6ds3tr_c_gy_flag_data_ready_get(struct pt *pt, stmdev_ctx_t *ctx, uint8_t *val))
 {
     lsm6ds3tr_c_status_reg_t *status_reg = (void *)&lsm6ds3tr_data_buffer[0];

     PT_BEGIN(pt);

-     lsm6ds3tr_c_read_reg(ctx, LSM6DS3TR_C_STATUS_REG, (uint8_t *)status_reg, 1);
+     lsm6ds3tr_c_read_reg(ctx, LSM6DS3TR_C_STATUS_REG, (uint8_t *)status_reg, 2);
     PT_WAIT_UNTIL(pt, ctx->read_write_done);
     *val = status_reg->gda;

     PT_END(pt);
 }
```

From what I can tell, it does not appear HAL_I2C_MemTxCpltCallback nor HAL_I2C_ErrorCallback gets called when it hangs.

**Expected behavior**
I would have expected this to work properly without the hacky change to read 2 byte instead.

dlech, 2022-10-25:
As you have probably seen in the code already, this chip seems to have some timing issues and has to have the bus reset occasionally. To fully understand the problem, the best thing to do would be to put a logic analyzer on the I2C bus to see what it is actually doing.

However, instead of polling for data ready, we should be able to use the INT1 interrupt pin that is connected to PB4 on the MCU. So maybe we can just avoid the problem that way?

davidwu226, 2022-10-25:
Thanks! I didn't know where INT1 was connected. This is the first time I'm hacking on a STM32, so just to be clear, you are suggesting I should try an interrupt driven approach using PB4, which would generate an interrupt on EXTI4, right?

dlech, 2022-10-25:
Yes, that is correct.

laurensvalk, 2022-10-25:
Nice! If you are going to try the interrupt driven approach, it would be great to look at the FIFO buffer: https://github.com/pybricks/support/discussions/675#discussioncomment-3304903

To get accurate integration, it would be beneficial to have equidistant time samples which the FIFO approach might provide. Then the math to integrate it can run asynchronously when there is enough data in the buffer to parse.

dlech, 2023-01-31:
I've implemented interrupts in https://github.com/pybricks/pybricks-micropython/commit/75dcf9479ab0217de6458afdcd65e5085644a2ba so this shouldn't be a problem any more since polling is not needed.

closed 2023-01-31 by dlech