pybricks/support #756: [Feature] Follow MicroPython in module naming, drop u-prefix?

laurensvalk, 2022-10-25:
Upstream MicroPython is likely dropping the `u` prefix in module names. For example, `usys` will just become `sys`. See https://github.com/micropython/micropython/issues/9018

For better or worse, it's probably worth discussing what we do before we release V3.2.

laurensvalk, 2022-10-25:
My vote is to just follow MicroPython, avoiding divergence.

dlech, 2022-10-25:
Dropping the prefix would also mean that people couldn't simply `pip install pybricks` to get code completion in VS Code anymore.

dlech, 2023-04-28:
One thing I've realized we could possibly do with our `pybricks` package is to use a `.pth` file to hide all of the "MicroPython" modules. Users would then have to manually configure their IDE to completely override where the intellesense looks for Python packages - which may not be technically possible in all IDEs.

romw314, 2025-04-14:
You could really alias these modules, and still keep the old names too. It would help me to port some Python modules to Pybricks. 😄