pybricks/support #757: [Bug] Autocomplete scans files that are not imported

laurensvalk, 2022-10-25:
**Describe the bug**

Everything from all files appears to be scanned, and they are all suggested for autocomplete.

**To reproduce**

For example, if you define:

```python
def my_useful_function(x, y):
    return x + y
```

in one file, it is suggested everywhere even if that module is not imported:

[image]

dlech, 2022-10-25:
This is a "feature" of the Monaco editor (there is probably an option turn it off). If there are no *real* code completions from Jedi, then the editor will just suggest anything that you have written before (this is what the "abc" icon means). For example `my_func` doesn't have to be a function. You can write `# my_func` and `my_func` will still be suggested.

laurensvalk, 2022-10-25:
Ah, that explains it.

Probably also explains why some terms are "remembered" even after you delete them.

> there is probably an option turn it off

Yeah, perhaps that's a good idea. I'm trying to think of cases where we'd miss it, but I can't think of any yet.