pybricks/support #759: [Bug] Available modules not suggested in auto-import

laurensvalk, 2022-10-25:
**Describe the bug**
This is the opposite of #757.

It seems that all files are scanned for their content, but not their name for importing.

[image]

Above, I would expect it to offer `my_module` as an import.

dlech, 2022-10-25:
This is due to https://github.com/pybricks/pybricks-api/blob/4573fd27609aae3bc7c6f24ed1384c926a735d95/jedi/src/pybricks_jedi/__init__.py#L260-L262

We can probably fix it by adding a check of [module_path](https://jedi.readthedocs.io/en/latest/docs/api-classes.html#jedi.api.classes.BaseName.module_path) so that filter only applies to the "system" directories and not the user directory.

laurensvalk, 2022-11-28:
Is it a bad idea to inject the names wherever its gets these from? (Would need update only when a file is deleted/added/renamed).

[image]

dlech, 2022-11-28:
I tried this a while ago and it wasn't as easy as it seems like it should be. I don't remember exactly where I was getting stuck, but I gave up for the time being and moved on to lower hanging fruit.

Another possible solution might be to figure out how to get Pyodide to install the `pybricks` package in a different location, then we could just change `sys.path` and not have to have any filter at all.

closed 2022-12-28 by dlech