pybricks/support #760: [Bug] beta.pybricks.com hangs on syntax error

BertLindeman, 2022-10-25:
**Describe the bug**
After bluetooth connect to a technic hub, clicking the "download and run" button, results in no reactions anymore on the webpage.

**To reproduce**
Steps to reproduce the behavior:
1. On win11 with Edge Go to 'beta.pybricks.com'
2. Load the program (added below after trying the reduce it to a minimum)
3. Click (or F5) the button 'download and run'
4. See that nothing works anymore
5. Only pressing CTRL-F5 revives the webpage.

**Expected behavior**
The syntax error to be reported as error

**remarks**
Obvious there is at least one syntax error also in this line: 
```python
motorB.run_target(TEST_SPEED, 0, wait=Falsee)
```
```Falsee``` should be ```False```

But the console shows an error on line 15:
[image]

There is a closing parens missing at that line.
After adding that `)` all works as expected.

**Test source**
```python
from pybricks.hubs import TechnicHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Button, Color, Direction, Port, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

# define motors:
motorA = Motor(Port.A, Direction.COUNTERCLOCKWISE)
motorB = Motor(Port.B, Direction.COUNTERCLOCKWISE)
steer = Motor(Port.C)
gearbox = Motor(Port.D)

if True:
    motorB.run_target(TEST_SPEED, 0, wait=Falsee)
    while ((motorA.angle() > 0) and (motorB.angle() > 0):
        if (motorA.angle() <= 1):
            motorA.stop()
        if (motorB.angle() <= 1):
            motorB.stop()
        wait(10)

```

laurensvalk, 2022-10-25:
Could this be the same as https://github.com/pybricks/support/issues/755?

BertLindeman, 2022-10-25:
Correct Laurens.

Did search for hang and such.. 

Is there a close-dup button available?

closed 2022-10-25 by dlech

BertLindeman, 2022-10-25:
Fixed in both Google Chrome and MS EDGE

Thank you, David, Laurens