pybricks/support #761: ``` Connection failed for unknown reason. ```

Render19, 2022-10-26:
at code.pybricks.com I try to install the firmware onto a technic hub but an error appears that says:
 "Unexpected error while trying to connect: Connection failed for unknown reason."

I'm on Windows 11 Home and use Google Chrome.
Ask if you need any more information.

I've looked all over your support page and tried **everything** I could but nothing works. Any Ideas?

dlech, 2022-10-26:
You could try https://beta.pybricks.com and see if it works better for you.

Beyond that, we will need log files as described in https://github.com/pybricks/support/discussions/270 to help troubleshoot.

Render19, 2022-10-26:
The beta sadly has the same error.

Here is the chrome device log:
[image]

dlech, 2022-10-26:
This looks like a problem between the Bluetooth chip and the Windows Bluetooth stack. You could try logging the Bluetooth packets with Wireshark as described in #270 to get a better look at what is going on.

It seems like Windows just has issues with some Bluetooth adapters. If you have another computer/chromebook/android device, you could try flashing the firmware with that instead. Sometimes, one the firmware is flashed, then you can use Windows to download and run programs. Trying a different Bluetooth adapter on Windows could be a possibility too.

Render19, 2022-10-26:
So I installed the firmware with a different computer but now I can't pair it to this computer. It still says "Connection failed for unknown reason"

And would you want to look at the bluetooth packet?

dlech, 2022-10-26:
Yes, I would be happy to look at it.

Render19, 2022-10-26:
This one is when I tried to install the firmware:
[Pybricks install firmware.zip](https://github.com/pybricks/support/files/9872979/Pybricks.install.firmware.zip)
And this one is just pairing:
[Pybricks Pair.zip](https://github.com/pybricks/support/files/9872981/Pybricks.Pair.zip)

dlech, 2022-10-26:
Is this with https://code.pybricks.com or https://beta.pybricks.com?

Render19, 2022-10-26:
https://code.pybricks.com/

dlech, 2022-10-26:
Can you try it with https://beta.pybricks.com/?

dlech, 2022-10-26:
Thanks for making all of the logs. They all look pretty much the same. The device connects, but Windows never tries to get the BLE services from the device. So, I'm afraid there isn't much we can do in Pybricks. 

If you have a different computer you can use, you could just use that. Or you might have better luck with a different Bluetooth adapter on your Windows computer that uses a different Bluetooth chip/driver.

Render19, 2022-10-26:
Ok thanks for all the help.

dlech, 2022-10-26:
For the record, would you mind sharing your Bluetooth Adapter driver info  from the Device Manager so we can see if other people with the same problem have the same Bluetooth chip/driver?

Mine looks like this:
[image]

Render19, 2022-10-26:
[image]