pybricks/support #766: [Bug] Bluetooth connection with 51515 hub not possible

BorisBrock, 2022-10-28:
I just downloaded the latest firmware via https://beta.pybricks.com/ to my 51515 hub.
After restarting the hub, the power button starts blinking blue, which is OK I guess.

Then I tried to connect my laptop via BT to the hub:

- The Hub appears in Manjaro's bluetooth settings as "Pybricks Hub" (OK)
- I can click on it to connect (OK)
- The LED on the hub now turns blue constantly (OK?)
- My laptop still shows that it is connecting for many seconds (not OK)
- After ~30s, the laptop gives up, and the LED on the hub starts blinking again (not OK)

PS:
After the above procedure, my laptop now sees two USB devices called "Pybricks Hub". Don't know what that means though...

PPS:
I did the same experiment with my Android smartphone. It shows exactly the same behavior as above (LED turns blue, phone keeps connecting, finally gives up. Now two BT devices are shown). Thus the problem seems not to be limited to my laptop.

laurensvalk, 2022-10-28:
There's no need to connect to the hub in your PC/Device settings.

Instead, it should just show up if you click the Bluetooth button in https://beta.pybricks.com/ 

Is the device list empty there?

BorisBrock, 2022-10-28:
Hitting the Bluetooth button directly gives me the following error:

> No Bluetooth adapter could be found.
> 
> Please connect or enable a Bluetooth Low Energy adapter and restart the browser.

laurensvalk, 2022-10-28:
So does your computer support Bluetooth low energy? If not built-in, a USB Bluetooth dongle should also work.

BorisBrock, 2022-10-28:
Hi, here are my results so far:

**Laptop**
- Brave Browser: Error, "No Bluetooth adapter found"
- Firefox: Error, "Web Bluetooth not supported"

**Android Smartphone**
- Firefox Mobile: "Web Bluetooth not supported"
- DuckDuckGo Privacy Browser: "Web Bluetooth not supported"

**Android Tablet**
- Firefox Mobile: "Web Bluetooth not supported"
- DuckDuckGo Privacy Browser: "Web Bluetooth not supported"
- Google Chrome: working! (yay)

Sadly Web Bluetooth seems to be a rather badly supported feature :-(

I would love to make it work on my laptop (Manjaro + Brave Browser) though...

laurensvalk, 2022-10-28:
Web bluetooth is supported in Chromium derivatives (Google Chrome, Microsoft Edge, or just Chromium).

That seems to include Brave, too.

So does it work if you enable Experimental Web Platform features on the `chrome://flags/`  ?

BorisBrock, 2022-10-28:
Yeah, I just installed Chromium. Here everything is working as expected right out of the box.
Thank you very much for your support!

PS: This is a really awesome project. Especially now that Lego officially killed Mindstorms, it feels great to have an alternative way to program my little creations :-)

closed 2022-10-28 by BorisBrock

laurensvalk, 2022-10-28:
Glad to hear it, have fun!

In case you're interested, we have a [projects repo](https://github.com/pybricks/pybricks-projects) with examples for all lego sets. Users can submit projects and they show up here: https://pybricks.com/projects/sets/

We try to add more examples, so feel free to submit something if you like :)