pybricks/support #772: [Bug] Jedi crash

laurensvalk, 2022-11-03:
It's still unclear how this happened, but I just noted this in the DevTools log:

```
preloading done.
python-worker.ts:148 Pyodide is ready.
sagas.ts:694 J: Traceback (most recent call last):
  File "/home/pyodide/pybricks_jedi/__init__.py", line 454, in complete
    completions = jedi.Script(code).complete(line, column - 1, fuzzy=True)
  File "/lib/python3.10/site-packages/jedi/api/helpers.py", line 487, in wrapper
    return func(self, line, column, *args, **kwargs)
  File "/lib/python3.10/site-packages/jedi/api/__init__.py", line 214, in complete
    return completion.complete()
  File "/lib/python3.10/site-packages/jedi/api/completion.py", line 170, in complete
    cached_name, completion_names = self._complete_python(leaf)
  File "/lib/python3.10/site-packages/jedi/api/completion.py", line 314, in _complete_python
    completion_names += self._complete_global_scope()
  File "/lib/python3.10/site-packages/jedi/api/completion.py", line 380, in _complete_global_scope
    completion_names += filter.values()
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 117, in values
    return self._convert_names(
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 114, in _convert_names
    return [self.name_class(self.parent_context, name) for name in names]
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 114, in <listcomp>
    return [self.name_class(self.parent_context, name) for name in names]
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 120, in <genexpr>
    for name in self._filter(
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 145, in _filter
    return list(self._check_flows(names))
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 156, in _check_flows
    check = flow_analysis.reachability_check(
  File "/lib/python3.10/site-packages/jedi/inference/flow_analysis.py", line 85, in reachability_check
    return _break_check(context, value_scope, first_flow_scope, node)
  File "/lib/python3.10/site-packages/jedi/inference/flow_analysis.py", line 93, in _break_check
    reachable = _check_if(context, check_node)
  File "/lib/python3.10/site-packages/jedi/inference/flow_analysis.py", line 120, in _check_if
    types = context.infer_node(node)
  File "/lib/python3.10/site-packages/jedi/inference/context.py", line 224, in infer_node
    return infer_node(self, node)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 156, in infer_node
    return _infer_node_if_inferred(context, element)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 169, in _infer_node_if_inferred
    return _infer_node_cached(context, element)
  File "/lib/python3.10/site-packages/jedi/inference/cache.py", line 44, in wrapper
    rv = function(obj, *args, **kwargs)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 174, in _infer_node_cached
    return _infer_node(context, element)
  File "/lib/python3.10/site-packages/jedi/debug.py", line 81, in wrapper
    return func(*args, **kwargs)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 82, in wrapper
    return func(context, *args, **kwargs)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 254, in _infer_node
    return infer_or_test(context, element)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 460, in infer_or_test
    types = context.infer_node(next(iterator))
  File "/lib/python3.10/site-packages/jedi/inference/context.py", line 224, in infer_node
    return infer_node(self, node)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 156, in infer_node
    return _infer_node_if_inferred(context, element)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 169, in _infer_node_if_inferred
    return _infer_node_cached(context, element)
  File "/lib/python3.10/site-packages/jedi/inference/cache.py", line 44, in wrapper
    rv = function(obj, *args, **kwargs)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 174, in _infer_node_cached
    return _infer_node(context, element)
  File "/lib/python3.10/site-packages/jedi/debug.py", line 81, in wrapper
    return func(*args, **kwargs)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 82, in wrapper
    return func(context, *args, **kwargs)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 197, in _infer_node
    value_set = context.infer_node(first_child)
  File "/lib/python3.10/site-packages/jedi/inference/context.py", line 224, in infer_node
    return infer_node(self, node)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 156, in infer_node
    return _infer_node_if_inferred(context, element)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 169, in _infer_node_if_inferred
    return _infer_node_cached(context, element)
  File "/lib/python3.10/site-packages/jedi/inference/cache.py", line 44, in wrapper
    rv = function(obj, *args, **kwargs)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 174, in _infer_node_cached
    return _infer_node(context, element)
  File "/lib/python3.10/site-packages/jedi/debug.py", line 81, in wrapper
    return func(*args, **kwargs)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 82, in wrapper
    return func(context, *args, **kwargs)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 184, in _infer_node
    return infer_atom(context, element)
  File "/lib/python3.10/site-packages/jedi/inference/syntax_tree.py", line 304, in infer_atom
    return context.py__getattribute__(atom, position=position)
  File "/lib/python3.10/site-packages/jedi/inference/context.py", line 45, in py__getattribute__
    names = self.goto(name_or_str, position)
  File "/lib/python3.10/site-packages/jedi/inference/context.py", line 34, in goto
    names = finder.filter_name(filters, name_or_str)
  File "/lib/python3.10/site-packages/jedi/inference/finder.py", line 36, in filter_name
    names = filter.get(string_name)
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 263, in get
    return [n for filter in self._filters for n in filter.get(name)]
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 263, in <listcomp>
    return [n for filter in self._filters for n in filter.get(name)]
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 109, in get
    return self._convert_names(self._filter(
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 145, in _filter
    return list(self._check_flows(names))
  File "/lib/python3.10/site-packages/jedi/inference/filters.py", line 156, in _check_flows
    check = flow_analysis.reachability_check(
  File "/lib/python3.10/site-packages/jedi/inference/flow_analysis.py", line 52, in reachability_check
    first_flow_scope = get_parent_scope(node, include_flows=True)
  File "/lib/python3.10/site-packages/jedi/parser_utils.py", line 245, in get_parent_scope
    if is_scope(scope):
KeyboardInterrupt

    at new_error (https://beta.pybricks.com/static/js/pyodide/0.21.3/pyodide.asm.js:1:185207)
    at https://beta.pybricks.com/static/js/pyodide/0.21.3/pyodide.asm.wasm:wasm-function[274]:0xdef7c
    at https://beta.pybricks.com/static/js/pyodide/0.21.3/pyodide.asm.wasm:wasm-function[275]:0xdf077
    at Module._pythonexc2js (https://beta.pybricks.com/static/js/pyodide/0.21.3/pyodide.asm.js:1:730490)
    at Module.callPyObjectKwargs (https://beta.pybricks.com/static/js/pyodide/0.21.3/pyodide.asm.js:1:112656)
    at Module.callPyObject (https://beta.pybricks.com/static/js/pyodide/0.21.3/pyodide.asm.js:1:112818)
    at y.apply (https://beta.pybricks.com/static/js/pyodide/0.21.3/pyodide.asm.js:1:118128)
    at Object.apply (https://beta.pybricks.com/static/js/pyodide/0.21.3/pyodide.asm.js:1:117541)
    at https://beta.pybricks.com/static/js/542.15209423.chunk.js:2:26327
    at p (https://beta.pybricks.com/static/js/542.15209423.chunk.js:2:3958)
    at Generator._invoke (https://beta.pybricks.com/static/js/542.15209423.chunk.js:2:3711)
    at Generator.next (https://beta.pybricks.com/static/js/542.15209423.chunk.js:2:4321)
    at n (https://beta.pybricks.com/static/js/542.15209423.chunk.js:2:9193)
    at c (https://beta.pybricks.com/static/js/542.15209423.chunk.js:2:9396)
    at https://beta.pybricks.com/static/js/542.15209423.chunk.js:2:9455
    at new Promise (<anonymous>)
    at https://beta.pybricks.com/static/js/542.15209423.chunk.js:2:9336
    at https://beta.pybricks.com/static/js/542.15209423.chunk.js:2:26736
(anonymous) @ sagas.ts:694
sagas.ts:131 Web Bluetooth is experimental on this platform. See https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md
sagas.ts:225 Downloading 360 bytes
```

dlech, 2022-11-03:
KeyboardInterrupt is normal to cancel an operation. We should be catching it with https://github.com/pybricks/pybricks-code/blob/6c373c302cfc9fae9302c1b245001b299b40fa33/src/pybricksMicropython/python-worker.ts#L33. 

Then it gets handled at https://github.com/pybricks/pybricks-code/blob/6c373c302cfc9fae9302c1b245001b299b40fa33/src/editor/sagas.ts#L682.

Maybe need to tweak something for a multiline regex match?

dlech, 2022-11-03:
I did some debugging and this seems to be working as expected. If this only happened in a development environment and we can't reproduce the issue, I think we can just assume the app was just in a bad state due to hot reloading and close the issue.

laurensvalk, 2022-11-03:
> If this only happened in a development environment 

I'm pretty sure this was on https://beta.pybricks.com, not localhost. But I don't mind closing it. I'll keep an eye out if it happens again.

laurensvalk, 2022-11-22:
> I think we can just assume the app was just in a bad state due to hot reloading and close the issue.

Closing until we see it again.

closed 2022-11-22 by laurensvalk