pybricks/support #773: [Bug] Use correct scale for hub renders in Pybricks Code

laurensvalk, 2022-11-03:
**Describe the bug**
The hub pictures render at a different scale.

[image]

**Suggested fix**
We could render the images with a fixed width or height so we don't have to make it complicated in Pybricks Code.

dlech, 2022-11-03:
> We could render the images with a fixed width or height so we don't have to make it complicated in Pybricks Code.

Would you mind doing this? The app is currently scaling them to 64x64 pixels.

laurensvalk, 2022-11-03:
Yeah, was already working on it, forgot to assign.

I'll just make them square.

We can export at 64x64 if you like as well, or anything else for that matter. I'll have a look at what comes out right.

dlech, 2022-11-08:
> If we find that the relative scale makes the small hubs look too small, I propose to scale down only Move Hub.

Currently looks like this (for anyone that wants to share their opinion):
[image]

closed 2022-11-08 by laurensvalk

laurensvalk, 2022-11-08:
Hopefully this helps a bit, otherwise feel free to re-open.