pybricks/support #779: [Bug] BLE Connection Problem with latest Pybricksdev ver a32

johnscary-ev3, 2022-11-06:
Hi,
Downloaded Pybricksdev a32 version package and command line today and also the latest primehub-firmware-build-2177.zip file.
The hub updated fine with pybricksdev command line a32 version.
Now I can use the hub fine with pybricks beta but with pybricksdev package I just get a timeout error.
It was working ok before I did the update. See below for errors and program.
Any ideas? I am using Python 3.10. Does that need to upgrade also?

```
PS C:\Users\johns\pybricksdev-demo> & c:/Users/johns/pybricksdev-demo/.venv/Scripts/python.exe c:/Users/johns/pybricksdev-demo/RemoteControl_Blast_New_June_2022.py
main: Start
Traceback (most recent call last):
  File "C:\Users\johns\pybricksdev-demo\.venv\lib\site-packages\bleak\backends\winrt\client.py", line 313, in connect
    await event.wait()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\asyncio\locks.py", line 214, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\johns\pybricksdev-demo\RemoteControl_Blast_New_June_2022.py", line 86, in <module>
    asyncio.run(main())
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\asyncio\base_events.py", line 649, in run_until_complete
    return future.result()
  File "c:\Users\johns\pybricksdev-demo\RemoteControl_Blast_New_June_2022.py", line 44, in main
    await hub.connect(device)
  File "C:\Users\johns\pybricksdev-demo\.venv\lib\site-packages\pybricksdev\connections\pybricks.py", line 199, in connect
    await self.client.connect()
  File "C:\Users\johns\pybricksdev-demo\.venv\lib\site-packages\bleak\backends\winrt\client.py", line 312, in connect
    async with async_timeout.timeout(timeout):
  File "C:\Users\johns\pybricksdev-demo\.venv\lib\site-packages\async_timeout\__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "C:\Users\johns\pybricksdev-demo\.venv\lib\site-packages\async_timeout\__init__.py", line 212, in _do_exit 
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
PS C:\Users\johns\pybricksdev-demo> 

import asyncio
import platform
import sys
 
try:
    import msvcrt
except ModuleNotFoundError:
    pass

try:
    import termios
    import tty
except ModuleNotFoundError:
    pass

from pybricksdev.connections.pybricks import PybricksHub, StatusFlag
from pybricksdev.ble import find_device

def read_key():
    if platform.system() == 'Windows':
        return msvcrt.getch()
    
    fd = sys.stdin.fileno()
    old_settings = termios.tcgetattr(fd)

    try:
        tty.setraw(fd)
        return sys.stdin.read(1)
    finally:
        termios.tcsetattr(fd, termios.TCSADRAIN, old_settings)

async def main():
    print('main: Start')

    hub = PybricksHub()

    device = await find_device()
    await hub.connect(device)
 
    try:
        forwarder_task = asyncio.create_task(forwarder(hub))
        try:
            # Added print_output, wait
            await hub.run('robot_Blast.py', wait=True, print_output=True)
        finally:
            forwarder_task.cancel()
    finally:
        # Disconnect from the hub
        await hub.disconnect()
        print('main: Stop')

async def forwarder(hub: PybricksHub):
    print("forwarder: Start")

    queue = asyncio.Queue()

    # wait for user program on the hub to start
    with hub.status_observable.subscribe(lambda s: queue.put_nowait(s)):
        while True:
            status = await queue.get()

            if status & StatusFlag.USER_PROGRAM_RUNNING:
                break

    print('forwarder: Hub Running')

    loop = asyncio.get_running_loop()
 
    # Keyboard command input loop
    while True:
        command = await loop.run_in_executor(None, read_key)

        # Try to send to the receiving hub
        try:
            await hub.write(bytes([ord(command)]))
        except:
            pass

#start it up
asyncio.run(main())

dlech, 2022-11-06:
If you are getting a timeout while connecting, this sounds like a Windows Bluetooth problem and not related to any Pybricks versions.

We would need to seem some Bleak logging and/or Bluetooth packet caputre to troubleshoot further. https://bleak.readthedocs.io/en/latest/troubleshooting.html

johnscary-ev3, 2022-11-07:
Ok I will try to do some logging to get more info.
However, seems like some problem with pybricksdev a32 since updated hub connects fine using Pybricks Beta environment now.
And it worked fine with pybricksdev a30 just before I upgraded to a32.
I tried to run my program using the pybricksdev a32 command line and got same timeout error. See below.
Seems like it can't find the hub now for some reason.

```
PS C:\Users\johns\pybricksdev-demo> pybricksdev run ble RemoteControl_Blast_New_June_2022.py
Searching for any hub with Pybricks service...
Traceback (most recent call last):
  File "C:\Users\johns\.local\pipx\venvs\pybricksdev\lib\site-packages\bleak\backends\winrt\client.py", line 313, in connect
    await event.wait()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\asyncio\locks.py", line 214, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "c:\users\johns\.local\bin\pybricksdev.exe\__main__.py", line 7, in <module>
  File "C:\Users\johns\.local\pipx\venvs\pybricksdev\lib\site-packages\pybricksdev\cli\__init__.py", line 384, in main
    asyncio.run(subparsers.choices[args.tool].tool.run(args))
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\asyncio\base_events.py", line 649, in run_until_complete
    return future.result()
  File "C:\Users\johns\.local\pipx\venvs\pybricksdev\lib\site-packages\pybricksdev\cli\__init__.py", line 205, in run
    await hub.connect(device_or_address)
  File "C:\Users\johns\.local\pipx\venvs\pybricksdev\lib\site-packages\pybricksdev\connections\pybricks.py", line 199, in connect
    await self.client.connect()
  File "C:\Users\johns\.local\pipx\venvs\pybricksdev\lib\site-packages\bleak\backends\winrt\client.py", line 312, in connect
    async with async_timeout.timeout(timeout):
  File "C:\Users\johns\.local\pipx\venvs\pybricksdev\lib\site-packages\async_timeout\__init__.py", line 129, in __aexit__
    self._do_exit(exc_type)
  File "C:\Users\johns\.local\pipx\venvs\pybricksdev\lib\site-packages\async_timeout\__init__.py", line 212, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
PS C:\Users\johns\pybricksdev-demo> pybricksdev -v
pybricksdev v1.0.0a32
PS C:\Users\johns\pybricksdev-demo>
```

dlech, 2022-11-07:
If it makes it to the `connect` method, this it is finding the hub, just not connecting. Enabling logging will tell is which part of the connecting process it is failing at.

johnscary-ev3, 2022-11-07:
Hi 
I collected two data log files. One when using Pybricks Beta to connect and the other when using Pybriksdev.
Not really sure what it means but the Beta one looks more successful since it actually connects.
Hope I did this correctly so you can get these files. Had to zip it for github.

[pybricksdevtest.zip](https://github.com/pybricks/support/files/9947760/pybricksdevtest.zip)

dlech, 2022-11-07:
Thanks for the logs. I'm able to reproduce the problem. It looks like a bug in the `bleak` library we use for BLE in Python triggered by the fact that we switched to random Bluetooth addresses in the Pybricks firmware. I can fix Bleak and then update Pybricksdev.

dlech, 2022-11-07:
Should be fixed in pybricksdev v1.0.0a33.

closed 2022-11-07 by dlech

johnscary-ev3, 2022-11-07:
Yes, it is working great now.
And the program download speed is about 2X faster now!
Thanks very much for quick help on this.