pybricks/support #788: Documentation not showing correctly on IntelliJ

laviRZ, 2022-11-11:
I'm not sure whether it's a pybricks or an IntelliJ problem, but only the parameters of the documentation shows when hovering over a function name.
[image]
[image]

dlech, 2022-11-11:
> IntelliJ

Isn't this a Java IDE? So I wouldn't expect it to work with Python. Or do you mean PyCharm? As far as I know, documentation follows Python standards and is picked up by Pyright/Pylance and Jedi so if it isn't getting picked up, it seems likely to be a problem with the IDE or whatever static analysis tool it uses.

laviRZ, 2022-11-12:
Nope, IntelliJ
They allow working with python with a plugin that makes it behave
functionally the same to PyCharm.

dlech, 2023-03-24:
Closing since we don't officially support IntelliJ/PyCharm.

Happy to answer additional questions at https://github.com/orgs/pybricks/discussions or review a pull request https://github.com/pybricks/pybricks-api to improve compatibility.

closed 2023-03-24 by dlech