pybricks/support #789: [Jerky acceleration on Beta]

9KuuCqmtebjTD, 2022-11-11:
Using this code below works fine in the standard version, but produces jerk acceleration on beta.

`from pybricks.pupdevices import Motor, Remote
from pybricks.parameters import Port, Direction, Stop, Button
from pybricks.tools import wait

# Initialize the motors.
steer = Motor(Port.C)
front = Motor(Port.A, Direction.COUNTERCLOCKWISE)
rear = Motor(Port.B, Direction.COUNTERCLOCKWISE)

# Lower the acceleration so the car starts and stops realistically.
front.control.limits(acceleration=1000)
rear.control.limits(acceleration=1000)

# Connect to the remote.
remote = Remote()

# Find the steering endpoint on the left and right.
# The middle is in between.
left_end = steer.run_until_stalled(-200, then=Stop.HOLD)
right_end = steer.run_until_stalled(200, then=Stop.HOLD)

# We are now at the right. Reset this angle to be half the difference.
# That puts zero in the middle.
steer.reset_angle((right_end - left_end) / 2)
steer.run_target(speed=200, target_angle=0, wait=False)

# Now we can start driving!
while True:
    # Check which buttons are pressed.
    pressed = remote.buttons.pressed()

    # Choose the steer angle based on the left controls.
    steer_angle = 0
    if Button.LEFT_PLUS in pressed:
        steer_angle -= 75
    if Button.LEFT_MINUS in pressed:
        steer_angle += 75

    # Steer to the selected angle.
    steer.run_target(500, steer_angle, wait=False)

    # Choose the drive speed based on the right controls.
    drive_speed = 0
    if Button.RIGHT_PLUS in pressed:
        drive_speed += 1000
    if Button.RIGHT_MINUS in pressed:
        drive_speed -= 1000

    # Apply the selected speed.
    front.run(drive_speed)
    rear.run(drive_speed)

    # Wait.
    wait(10)`

laurensvalk, 2022-11-11:
Can you describe which model you used?

Can you reproduce with a smaller script? With just one motor and one hub?

laurensvalk, 2022-12-07:
The actionable item here is to do some basic tests with the Technic L and XL motors to check for any unusual behaviors, and take action if needed.

These motors haven't been tested extensively.

laurensvalk, 2022-12-09:
It seems to work OK on the model too. I'll close it for now. We can re-open when we have more information.

laurensvalk, 2022-12-09:
Perhaps the acceleration curve isn't quite as linear as it could be. This could be worth comparing to previous releases in a log.

laurensvalk, 2022-12-12:
This is more of a nice-to-have performance optimization than an immediate issue. We can still re-open for future evaluation.

laurensvalk, 2023-02-23:
Current state of affairs on latest (https://github.com/pybricks/pybricks-micropython/commit/830579b255b526779c1ec29c20dd4286bdff9080).

[image]
[image]

```python
from pybricks.pupdevices import Motor
from pybricks.tools import wait
from pybricks.parameters import Port, Direction
from pybricks import version

print(version)

# Initialize the fronts.
steer = Motor(Port.C)
front = Motor(Port.A, Direction.COUNTERCLOCKWISE)
rear = Motor(Port.B, Direction.COUNTERCLOCKWISE)

# Allocate the data logs.
DURATION = 8000
DIV = 6
front.log.start(DURATION, DIV)
front.control.log.start(DURATION, DIV)

# Run the front.
front.run(1000)
rear.run(1000)
wait(2000)
front.run(0)
rear.run(0)
wait(2000)
front.run(-1000)
rear.run(-1000)
wait(2000)
front.run(0)
rear.run(0)
wait(2000)

# Transfer data logs.
print("Transferring data...")
front.log.save("servo.txt")
front.control.log.save("control.txt")
print("Done")

```

laurensvalk, 2023-02-23:
And it was like this on https://github.com/pybricks/pybricks-micropython/releases/tag/v3.2.3

So it has already become a bit better with the recent updates to friction models.

[image]
[image]

laurensvalk, 2023-02-23:
If we include OP's acceleration value (1000) it is a bit worse. The speed graphs don't look terrible but the oscillations in the feedback signal is what you hear/observe:

[image]

laurensvalk, 2023-02-23:
But is was indeed better in V3.1 as stated by OP:

[image]

laurensvalk, 2023-02-23:
May 2022, https://github.com/pybricks/pybricks-micropython/commit/2206d82488bbf9a38186d4259c8d33761f700328

[image]

laurensvalk, 2023-02-23:
27 February 2022, https://github.com/pybricks/pybricks-micropython/commit/81d342d2a6714159fdb123e07e74b5b070cf6300

[image]

laurensvalk, 2023-02-23:
21 December 2021, https://github.com/pybricks/pybricks-micropython/commit/d7534640fec7fddb67d6a121d9038548d7bba6ce

[image]

laurensvalk, 2023-02-23:
25 jan 2022: https://github.com/pybricks/pybricks-micropython/commit/e07adde528f4eaca1c1f73996d86ec9e0c78edc1

[image]

laurensvalk, 2023-02-23:
Binary search says January 2022. To be continued!

laurensvalk, 2023-02-24:
18 jan 2022: https://github.com/pybricks/pybricks-micropython/commit/688a694b8930df83b43d9253af4737610c133f66, so second half of January it is.

[image]

laurensvalk, 2023-02-24:
later on 18 jan 2022: https://github.com/pybricks/pybricks-micropython/commit/8cf354aefcc965787594074cec154ec52275b02c

[image]

laurensvalk, 2023-02-24:
later on 18 jan 2022: https://github.com/pybricks/pybricks-micropython/commit/56d92b8876180dc66bac117bcb419d7d59413b76

[image]

> But is was indeed better in V3.1 (16 dec 2021) as stated by OP, despite still using the same PID gains...

So this turns out to be incorrect. It *was* the PID gain change.

laurensvalk, 2023-02-24:
But even with the lower gains on the latest firmware, it still isn't as smooth as it was:

[image]

laurensvalk, 2023-02-24:
Adjusted gains as above, 15 july 2022: https://github.com/pybricks/pybricks-micropython/commit/4535e36cbbd228cc5bbea546d5f51c7d93f3914e, still smooth, so try more recent.

laurensvalk, 2023-02-24:
Adjusted gains as above, 21 oct 2022: https://github.com/pybricks/pybricks-micropython/commit/c2615ce3bf7bf6ad4d95fe53ff630182f87347c2, still smooth, so try more recent.

[image]

laurensvalk, 2023-02-24:
Adjusted gains as above, 11 nov 2022: https://github.com/pybricks/pybricks-micropython/commit/be3beb49305f92eac146078ff214667e96a8ec7d, still smooth, so try more recent.

[image]

laurensvalk, 2023-02-24:
Adjusted gains as above, 6 feb 2023: https://github.com/pybricks/pybricks-micropython/commit/feb84b3556eb452cf8763c77768400d6213b8119, still smooth, so try more recent.

[image]

laurensvalk, 2023-02-24:
Adjusted gains as above, 6 feb 2023: https://github.com/pybricks/pybricks-micropython/commit/a8fcac09d3135978e4de8dc760e3a72b29b919e8. So the (new) oscillations at low KD gains are due to increased observer feedback:

[image]

laurensvalk, 2023-03-17:
This is fixed by https://github.com/pybricks/pybricks-projects/pull/84

closed 2023-03-17 by laurensvalk