pybricks/support #790: [Feature] remember the path a source file was imported, to save it back there (beta.pybricks.com)

BertLindeman, 2022-11-11:
**Is your feature request related to a problem? Please describe.**
In beta.pybricks.com I use to have a number of sources loaded. Not all imported from the same location.
After an update I like to store it back where it came from.
Would be nice (efficient) if the save dialog started in the directory where the source originally came from.

**Describe the solution you'd like**
Let the save dialog start in the location where **this** source was loaded form.

**Describe alternatives you've considered**
Sometimes, I paste the source location in the source.

**Additional context**
It **is** nice that the dialog ```import a file``` starts at the last used location.
And not always in ```documents```

laurensvalk, 2022-11-11:
I'm not sure the app can see where you uploaded it from, for security reasons (which is probably a good thing).

As far as I know, Pybricks is very stable about keeping your files in the app, I've never seen anything gone missing.

For more advanced file management, I think we would encourage you to set up something with your local editors combined with Pybricksdev.

BertLindeman, 2022-11-11:
Thanks Laurens,

I will encourage myself to paste the location in the source (Notepad++ does it easily)
Just to help my memory.
I use pybricks on windows and in the LEGO-room on Linux, so the two **might** have a different version of the same source.
Some more discipline I need to apply to myself ;-)

dlech, 2023-03-07:
I'm going to go ahead and close this one since I don't think it is technically possible. As an alternate solution, we could consider adding support for alternate storage such a Google Drive in the future.