pybricks/support #793: remote not paring to hub after latest update

tdamsma, 2022-11-11:


dlech, 2022-11-11:
Can you please share the exact firmware version and an example program to reproduce the problem?

tdamsma, 2022-11-11:
The firmware is whatever is pushed as latest on beta.pybricks.com as of today. The example program to reproduce is just call `Remote()` to initiate a pairing. I tried also with `Remote(timeout=None)` and that was just stuck.

dlech, 2022-11-11:
We just published a new version today, so depending on what time of day it could be a different firmware version. So without an exact version number, we can only guess which version is being used. You can find the version here:
[image]

 or `from pybricks import version; print(version)`.

With Pybricks firmware v3.2.0b5 on a Technic hub and the hub connected to Pybricks Beta on Linux, I am able to get the remote connected to the hub.

Which hub and operating system are you using? Do you turn on the remote before of after you start the program on the hub?

dlech, 2022-11-11:
Oops, I should read more carefully, I see that you said the BOOST Move hub already.

dlech, 2022-11-11:
I'm able to reproduce the problem with a move hub, so we will look into it.

dlech, 2022-11-11:
It appears to be working when not connected to Pybricks Beta:

Steps:
- download and run program with Pybricks Beta (doesn't work but needed to copy program to hub)
- disconnect hub from Pybricks Beta
- press button on hub to start program
- press button on remote to connect remote

```python
from pybricks.hubs import MoveHub
from pybricks.pupdevices import Motor, ColorDistanceSensor, Remote
from pybricks.parameters import Button, Color, Direction, Port, Stop
from pybricks.robotics import DriveBase
from pybricks.tools import wait, StopWatch

hub = MoveHub()

remote = Remote()

hub.light.on(Color.GREEN)
remote.light.on(Color.GREEN)

while True:
    print(remote.buttons.pressed())
    wait(1000)
```

dlech, 2022-11-11:
Found and fixed the problem. You can use [this firmware](https://github.com/pybricks/pybricks-micropython/suites/9262937254/artifacts/433639636) until the next beta release.

closed 2022-11-11 by dlech

tdamsma, 2022-11-11:
Well that was quick! Won't be able to test until Monday