pybricks/support #794: EV3 giving OSError: [Errno 5] even though hardware is tested and connected

Intergalactic-Carpet, 2022-11-12:
I'm making a simple tank movement program and all the code works correctly but a few seconds after the program starts running, it gives an OSError: [Errno 5]
Here is the code if it turns the problem is the code
```python
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile
from time import sleep
tank_speed = 0
tank_turn = 0
ev3 = EV3Brick()
sleep(0)
# HARDWARE SETTINGS
remote_frequency = 1
left_motor_port = Port.A
right_motor_port = Port.D
infrared_sensor_port = Port.S3

def restrict(in_, min=-1, max=1):
    if in_ > max:
        in_ = max
    elif in_ < min:
        in_ = min
    return in_

def start():
    global tank_speed, tank_turn
    ev3.speaker.beep()
    while True:
        data = InfraredSensor(infrared_sensor_port).buttons(remote_frequency)
        inc = 0.1
        if len(data) > 0:
            for _ in range(len(data)):
                if data[_] == Button.LEFT_UP:
                    tank_speed += inc
                elif data[_] == Button.LEFT_DOWN:
                    tank_speed -= inc
                elif data[_] == Button.RIGHT_UP:
                    tank_turn -= inc
                elif data[_] == Button.RIGHT_DOWN:
                    tank_turn += inc
                elif data[_] == Button.BEACON:
                    if False:
                        if InfraredSensor(infrared_sensor_port).beacon(1)[0] > 30:
                            angle = InfraredSensor(infrared_sensor_port).beacon(1)[1]
                            tank_turn = angle / 10
                            if angle > -10 and angle < 10:
                                tank_speed = InfraredSensor(infrared_sensor_port).beacon(1)[0] / 5
                        else:
                            tank_turn = 0
                            tank_speed = 0
                    else:
                        tank_turn = 0
                        tank_speed = 0
        Motor(left_motor_port).dc(restrict(tank_speed + tank_turn) * 100)
        Motor(right_motor_port).dc(restrict(tank_speed - tank_turn) * 100)

start()

```

laurensvalk, 2022-11-12:
These bits of code in your example keeps re-initializing the sensors and motor every time. Eventually the robot will give an error as you run out of resources.
```python
while True:
    # .... other code here ....
    InfraredSensor(infrared_sensor_port).beacon(1)
    # .... other code here ....
    Motor(left_motor_port).dc(restrict(tank_speed + tank_turn) * 100)
    Motor(right_motor_port).dc(restrict(tank_speed - tank_turn) * 100)
```

Instead it is better do so something like this:

```python
# Initialize (and reset) the devices only once at the start
ev3 = EV3Brick()
left_motor = Motor(Port.A)
right_motor = Motor(Port.D)
ir_sensor = InfraredSensor(Port.S3)

# Now you can use them in the rest of your code as much as you want
while True:
    # .... other code here ....
    ir_sensor.beacon(1)
    # .... other code here ....
    left_motor.dc(restrict(tank_speed + tank_turn) * 100)
    right_motor.dc(restrict(tank_speed - tank_turn) * 100)

```

Intergalactic-Carpet, 2022-11-12:
Thankyou

closed 2022-11-12 by Intergalactic-Carpet