pybricks/support #801: [Feature] Access RSSI (Received Signal Strength Indicator) from nearby hubs (or other devices)

pbochynski, 2022-11-13:
The recent [implementation of broadcast](https://github.com/pybricks/pybricks-micropython/pull/80) gives access to advertising data from nearby devices. It would be great if I could access the latest RSSI data from discovered devices. 

Usually, there is plenty of BLE devices around, so reading should be limited to subscribed names/ids. Alternatively, we could get RSSI data along with `Broadcast.receive`.

I know that RSSI data is not accurate and can be distorted by absorption, interference, or diffraction but could be used to calculate a distance from another hub or device. My idea is to build a return-to-home function or even a kind of indoor navigation using triangulation (3+ devices nearby).

laurensvalk, 2022-11-14:
Thanks for opening this issue. I was chatting about this with @nstrijbosch last week.

We thought it might be useful to add an `info(topic)` method to get additional info about signals. This could perhaps return a tuple or dictionary of:

- RSSI
- Timestamp (so you can choose to ignore old messages)
- Received counter
- Anything else?

A single `info()` method is still fairly minimal and would keep the `receive` API simple.

pbochynski, 2022-11-15:
Yes - info method would be perfect.

NStrijbosch, 2022-12-18:
https://user-images.githubusercontent.com/67324303/208315652-74a0d290-aa9a-446d-ac63-b029bbe3e3d0.mp4

A quick test where the inventor hub reads the RSSI value corresponding to the advertising data of the technic hub. It could be pretty usable to distinguish between being close or far away. The values do not seem stable enough such that they can be used to estimate distance.

laurensvalk, 2022-12-18:
Awesome!

Good use for the tiles :smile:

NStrijbosch, 2022-12-27:
> It could be pretty usable to distinguish between being close or far away. The values do not seem stable enough such that they can be used to estimate distance.

To visualize this argument a plot of the distance vs RSSI value. This was an experiment with a city hub that advertises data and a primehub that measures the distance with the UltrasonicSensor and reads the RSSI values:
[image]

pbochynski, 2022-12-28:
Definitely, it is not accurate, but the trend is visible. I think it still can be useful for larger distances and slow movements. I mean aggregating several measurements, e.g. min(RSSI) of 10 measurements should give better estimates (like <0.5m, ~1m, ~2m, >3m). Thanks for updating the PR - I will do some experiments with it.

dlech, 2023-04-14:
Another alternate with an API to get RSSI is at https://github.com/pybricks/pybricks-micropython/pull/158.

closed 2023-05-16 by dlech