pybricks/support #802: [Feature, Bug] Remote close method.

laurensvalk, 2022-11-14:
**Is your feature request related to a problem? Please describe.**
This was proposed in https://github.com/pybricks/pybricks-micropython/commit/a2b2e44eea37ee994204b662302134c5e4bcfb73 but never merged, so I had a look at how well it works. 

**Describe the solution you'd like**
With `close()` included, you can quite easily make programs that allow the remote to hop between hubs.

For example, take a Prime Hub and a Technic Hub, each with a motor on Port A.

They both run this program, modified on the lines indicated with `<----`

```python
from pybricks.hubs import PrimeHub
from pybricks.pupdevices import Remote, Motor
from pybricks.parameters import Button, Color, Port
from pybricks.tools import wait

# Replace this with your hub. <----
hub = PrimeHub()

# Replace this with a unique color <----
pair_color = Color.GREEN

# Initialize a motor.
motor = Motor(Port.A)

while True:

    # Blink to indicate we want to connect.
    hub.light.blink(pair_color, (200, 200))

    # Try to connect, and retry on timeout.
    remote = Remote(timeout=None)

    # Light solid to indicate we are connected.
    remote.light.on(pair_color)
    hub.light.on(pair_color)

    while True:

        # Wait until a button is pressed.
        pressed = remote.buttons.pressed()

        if Button.CENTER in pressed:
            break
        elif pressed:
            motor.dc(50)
        else:
            motor.stop()
        
        # Wait some time.
        wait(10)

    # Disconnect.
    hub.light.on(Color.YELLOW)
    motor.stop()
    remote.close()

    # Give other hub a chance to connect.
    wait(2000)
```

This uncovers a few interesting Bluetooth problems that may help us identify underlying bluetooth bugs seen elsewhere.

This **only** works well if the Prime Hub **is** connected to Pybricks Code. This was new to me.

When not connected, we get all sorts of strange conditions happening.

So my vote is not to merge this for 3.2 but include it in the next big Bluetooth release (3.3 or 3.4).

laurensvalk, 2022-11-14:
https://user-images.githubusercontent.com/12326241/201695626-a0ca0dea-e89d-418b-8b29-7a26b9af3698.mp4

BertLindeman, 2022-11-14:
Laurens,

would using ThisHub make this program easier to be used on different hubs?

Or is ThisHub not to be promoted?

(You lack "<----" on the PrimeHub import above)

laurensvalk, 2022-11-14:
We're not advocating for it due to the difficulties we have seen in ev3dev. One class to fit all sounds good, but often leads to confusion over time as functionality diverges.

It also gives problems with autocomplete, since it wouldn't know which hub it is.

BertLindeman, 2023-02-19:
While thinking about [discussions entry #975](https://github.com/orgs/pybricks/discussions/957)

> **Is your feature request related to a problem? Please describe.**
> This was proposed in [pybricks/pybricks-micropython@a2b2e44](https://github.com/pybricks/pybricks-micropython/commit/a2b2e44eea37ee994204b662302134c5e4bcfb73) but never merged, so I had a look at how well it works.

Have been looking through "Actions" in [github.com/pybricks/pybricks-micropython/actions](https://github.com/pybricks/pybricks-micropython/actions?page=17&query=actor%3Adlech+is%3Asuccess+workflow%3ABuild) 
And found [pybricks.pupdevices.Remote: add close() method #1221](https://github.com/pybricks/pybricks-micropython/actions/runs/1144437511)

Is that the firmware you refer to, Laurens?
That firmware is expired, so cannot download now.
But as it happens I DID download that firmware from 2021-08-18.
But of course that firmware is now incompatible with code.pybricks.com
Looks like a dead end for me at the moment.

laurensvalk, 2024-02-19:
This was also requested in https://github.com/orgs/pybricks/discussions/1445

closed 2024-04-04 by laurensvalk