pybricks/support #806: pybricks.eth button in sponsors dialog does nothing

dlech, 2022-11-18:
**Describe the bug**
There is a `pybricks.eth` button in the sponsors dialog, but clicking it does not do anything.

**To reproduce**
Steps to reproduce the behavior:
1. Go to beta.pybricks.com
2. Click on the heart icon
3. Click the pybricks.eth button
4. Nothing happens

**Expected behavior**
It should link to an external page like the other buttons.

**Screenshots**
[image]

laurensvalk, 2022-11-19:
This was intentional. It's really just intended as an option for people who know how this works.

It doesn't need to be a button actually.

But if we want to make it nicer, we could make it copy that domain name to the clipboard.

dlech, 2022-11-19:
It would be nice if it linked to a page or had a pop up that explained what it is for those who don't know how it works.

closed 2022-11-25 by dlech