pybricks/support #807: [Bug] With two pybricks code windows open, file menu opens on both

laurensvalk, 2022-11-19:
**Describe the bug**
Open two browser tabs or windows. Click the left hand file icon to open the file bar. Now it opens on both browser windows.

**Expected behavior**
Button activity constrained to one browser window.

dlech, 2022-11-19:
The root cause is that we use local storage to persist the state which is shared between all windows. There is quite a bit more state than just the selected activity tab that is tied to local storage and will affect all open windows.

- The selected hub for the hub picker control.
- The use template selection in the new file dialog.
- The state of the advanced section of the flash Pybricks firmware dialog.
- The show/hide docs state.

The position of panes that control the size of the docs and the terminal also use local storage, but the control doesn't update when the storage changes so we don't see the change in other windows.

The oneshot for showing the Pybricks tour on first run also uses local storage, but in that case we do want the state shared between windows.

dlech, 2022-11-19:
> **Expected behavior**
> Button activity constrained to one browser window.

What is the expected behavior when you close all windows that are in different states and then open a new window?

laurensvalk, 2022-11-19:
I think the current behavior is definitely fine for this release.

I was giving a demo on a small screen when I noticed this, but it is easy enough to just close the file menu again.

dlech, 2022-12-06:
The cases of the selected activity tab and the documentation show/hide state have been fixed.

closed 2022-12-06 by dlech