pybricks/support #812: Raspberry Pi Build Hat support

fc3sunfire, 2022-11-26:
From what I can gather and read, the Raspberry Pi itself seems supported with the use of the Chromium browser, but are you gents looking at maybe natively on the Raspberry Pi and use the Build Hat as like a hub of its own?

I am a bit of a coder myself, so I am checking out how to contribute and would love to work with you guys to get this up and running.

dlech, 2022-11-26:
From what I understand, the main problem is that the firmware for the Build Hat has to be signed and we don't have the signing key ([related code](https://github.com/RaspberryPiFoundation/python-build-hat/blob/16f50efcdaf0ba8593f2d09f369bf6539d63ada0/buildhat/serinterface.py)).  

There was also some related discussion in https://github.com/RaspberryPiFoundation/python-build-hat/issues/135.

So for now, we can't do anything until the RPi foundation changes their minds about allowing 3rd party firmware. They have made the comment in the past that they are potentially open to the idea if there is enough demand, so keep asking them if this is something you really want.

fc3sunfire, 2022-12-08:
Also, had a couple things, one: https://snapcraft.io/install/micropython/raspbian
is a MicroPython on Raspberry Pi, can we use this to port over the PyBricks

Also I was thinking, do we need to have PyBricks replace the firmware? I say this because of all the capabilities of the Pi. I am sure there is a performance drag, but can we do something on top of the OS? I just found this website I posted above, so going to see what possibilities this may open or not.

laurensvalk, 2022-12-08:
Theoretically, you could indeed just make a wrapper API that makes it look like Pybricks and make it call the respective Build Hat command underneath. But it wouldn't work well because you'd keep all the [slow performance of the build hat](https://www.youtube.com/watch?v=ZY8wgRCnZYQ&t=302s) . You would also not have features like stall detection, see https://github.com/RaspberryPiFoundation/python-build-hat/issues/135.

When it comes to replacing the firmware on the hat, we wouldn't necessarily put pybricks-micropython on it (although you could). Instead, we could put the pybricks-pbio components in. That's all the drivers we wrote for the LEGO devices, so without MicroPython. A hybrid of those approaches could work too.

romw314, 2025-06-26:
This is interesting, but I didn't try it myself: https://github.com/muzkr/hachi

laurensvalk, 2025-07-14:
<sup>Originally posted by @gregorianrants July 14, 2025</sup>

> it was announced a few days ago that pi Build HAT firmware now fully open source
> 
> the article can be found here https://www.raspberrypi.com/news/build-hat-firmware-now-fully-open-source/
> 
> the git hub repo can be found here https://github.com/raspberrypi/buildhat
> 
> in a previous ideas discussion i asked Is there any interest in making a Pybricks for the buildhat? https://github.com/orgs/pybricks/discussions/1960
> 
> lack of a signing key was cited as the main technical barrier to being able to do this. Does this mean pybricks on the build hat is now at least possible? i realise even if it is possible it would also take a fair bit of programming effort and time that the team may not have.
> 
> i decided to make a seperate post for this news as it is pretty big news.

laurensvalk, 2025-07-14:
Technically, it should now be possible.

As we've indicated elsewhere, we would still be interested to discuss with the Raspberry Pi Foundation about creating something like this, but there haven't been any conversations about this so far.

laurensvalk, 2025-07-14:
Thanks! I think the Build HAT could align with that initiative really well. (But it would work a lot better when working together with their team, so we don't end up with an half-finished, unofficial version that just splits support rather than make it better.)

Thinking outside the box, it would work really well with the Raspberry Pi on its own too. If you think about it, you could just as well attach your SPIKE Prime to the Raspberry Pi with a USB cable.

gregorianrants, 2025-07-14:
I have contemplated various combinations with the pi. i dont have spike but do have ev3 and technic hub. 

The beauty of the build hat though is that it is £24 and you can power the build hat and pi from 6 AA batteries.  My reason for getting the build hat was to build a mobile robot. for which this is great.  I tend to use my hubs for stationary or lego only projects.

Unfortunately the pid on the build hat is near useless, it actually sends motor in the wrong direction when you start them.  I managed to come up with a way of doing the pid in python and just reading encoders and setting with pwm, and got my mobile robot working.  I am happy with this solution just now.

here is the github for the project
https://github.com/gregorianrants/composed-robot

Pybricks on the build hat would just be awesome if it ever came about.  I totally get that you only want to do it if it can be done properly though and are busy with a lot of work on the current libraries.

If i had know that it was going to be so diffiucult to get the build hat working i would probably have just went for a hub pi combo and I dont think it is a bad idea. the ev3 motors are pretty powerfull and could probably handle some extra batteries for the pi.

davelongdev, 2025-10-26:
Any news on getting going with pybricks for the build hat?  Seems like it could be an inexpensive way for people to use pybricks?

dlech, 2025-10-26:
We've been super-busy working on EV3 support lately. And this is lower on priority list. Hopefully some day it will get done.

I did buy one of these recently, so if anyone wants to have a go at trying to get things started, I'm always happy to answer questions and review pull requests.

davelongdev, 2025-10-26:
I would like to work on it, but not sure how to get started at the moment.  Any thoughts on how to get going?

davelongdev, 2025-10-26:
I see.  It seems like that's going to require a bit of work.  Thanks for all the info.

I'll get started on trying to build the official firmware, start playing around with the build hat, and see how it all goes from there.

By the way, my main use case for this is to give some of the kids on our FLL teams who don't have lego hubs an inexpensive way to practice with pybricks at home, but I also think others would benefit as well.

One question - would we need to attach a build hat to a raspberry pi to use pybricks, or would we be able to use the built hat on its own?

dlech, 2025-10-26:
> raspberry pi zero

As long as you don't mind soldering on the pins yourself. 😄

davelongdev, 2025-11-26:
Just an update - I've made a lot of progress on this project. There is still a lot to do though. I'll post some more info on the status when I have a chance.  I'm glad there are people  interested in this!  If anyone wants to hop on zoom some time to collaborate, let me know.

davelongdev, 2025-11-27:
Ok - So here's what I've done so far -

---

1 -  Build the buildhat firmware from source.  

I was able to generate the image file and signature file by following the instructions in the repo.  I had to create and build it in an ubuntu vm, as I couldn't get it working on arch.  I ran into a few complications here.  

First, I needed to install the pico-sdk.  https://github.com/raspberrypi/pico-sdk

The pico-sdk required some dependencies.  For ubuntu, I ran 

```sudo apt install cmake python3 build-essential gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib```

I also had to initialize the tinyUSB module in order to avoid some issues.  To do this, I had to cd into the pico-sdk directory and run ```git submodule update --init```

I also had to install pico-tool.

For this, I had to download the prebuilt binary.

https://github.com/raspberrypi/pico-sdk-tools/releases

I then had to unzip with ```tar -xvf` <filename>``` and then set and environment variable for pico tool with ```export picotool_DIR=path/to/dir```

---

2) Conduct Research

I found that there is some good info about the build hat in these two repos - 

https://github.com/muzkr/buildhat_mp

https://github.com/muzkr/hachi

They contain helpful info about how to program the flash using two solder points as SWD connections, as well as info about the board's components.

---

3) Access the board through the SWD solder points and dump the contents of the existing flash (which appears to contain just the build hat bootloader or BHBL) using the raspberry pi debug probe, openOCD, and telnet.  

I also discovered the following:

 - it appears that the build hat has 2MB of flash memory which begins at address 0x10000000
 - the total amount of flash in bytes for 2MB should be 2097152
 - rp 2040 seems to have 264kb of on-chip SRAM in 6 banks starting at 0x20000000 and 16kB read-only memory (ROM) at address 0x00000000. 

https://www.adafruit.com/product/5699

https://openocd.org/

The following Commands were useful.

```
sudo openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000" (starts openocd server)
telnet (starts telnet)
open localhost 4444 (connects to openocd server)
dump_image `<filename>` 0x10000000 2097152 (openocd command)

```

---

Some of the next steps I was thinking of were

1)  try to flash the image that I dumped from the build hat onto another build hat to verify that the BHBL was successfully copied

and

2) try to install and test the hachi bootloader and use it to load a simple program to test that it works.

Now, I also plan to look over the work that @dlech did recently and see how it relates to what I've been working on.

---

Hope this helps out a bit.