pybricks/support #813: [Bug] Only RED color works with remote.light.on(color)

pbochynski, 2022-11-26:
**Describe the bug**
I have built firmware from the master branch and I executed the code for changing the color of the remote led:
https://beta.pybricks.com/static/docs/v2.4.0/pupdevices/remote.html#changing-the-light-color-using-the-buttons
It looks like only red one work (different brightness of red work), but green or blue do not work at all. With Color.BLUE the light is off.

**To reproduce**
Just follow the [contributing guide](https://github.com/pybricks/pybricks-micropython/blob/master/CONTRIBUTING.md)

**Expected behavior**
All colors work :)

**Screenshots**
https://youtu.be/qk_zqtszHfs

laurensvalk, 2022-11-26:
What color does it blink prior to connecting?

dlech, 2022-11-26:
I just tested building from commit 7a7504ec and it is working for me (using technic hub and the linked python script).

pbochynski, 2022-11-26:
It is blinking white before I connect it. With standard firmware (from beta.pybrics.com) it works well. This is a firmware I've built:
[firmware-master.zip](https://github.com/pybricks/support/files/10097093/firmware-master.zip)

dlech, 2022-11-26:
I tried the firmware.zip you shared and see the problem as well.

What compiler are you using?

pbochynski, 2022-11-26:
I installed ArmGNUToolchain 11.3.rel1.

dlech, 2022-11-26:
Can you zip up and share the contents of `bricks/technichub/build/ `?

pbochynski, 2022-11-26:
Here you are:
[technichub-build.zip](https://github.com/pybricks/support/files/10097128/technichub-build.zip)

dlech, 2022-11-26:
Thanks. Apparently, that version of the compiler is optimizing something out of `pb_type_pupdevices_Remote_light_on` that it shouldn't be. In your `firmware.elf.map`, `pb_type_pupdevices_Remote_light_on` is 0x50 bytes but in mine, it is 0x78 bytes.

pbochynski, 2022-11-26:
Thanks for the quick diagnosis. I will try other versions (new beta and older version) and let you know if I get 0x78 there. Any other suggestions on what should I try?

dlech, 2022-11-26:
Disassembly shows it is the lines ending in `* 3 / 8` are being optimized to `= 0`. So perhaps we need a cast to int there.

```diff
diff --git a/pybricks/pupdevices/pb_type_pupdevices_remote.c b/pybricks/pupdevices/pb_type_pupdevices_remote.c
index 9983523e..31ca6aa3 100644
--- a/pybricks/pupdevices/pb_type_pupdevices_remote.c
+++ b/pybricks/pupdevices/pb_type_pupdevices_remote.c
@@ -117,8 +117,8 @@ STATIC void pb_type_pupdevices_Remote_light_on(void *context, const pbio_color_h
 
     // The red LED on the handset is weak, so we have to reduce green and blue
     // to get the colors right.
-    msg.payload[1] = msg.payload[1] * 3 / 8;
-    msg.payload[2] = msg.payload[2] * 3 / 8;
+    msg.payload[1] = (uint32_t)msg.payload[1] * 3 / 8;
+    msg.payload[2] = (uint32_t)msg.payload[2] * 3 / 8;
 
     pbdrv_bluetooth_write_remote(&remote->task, &msg.value);
     pb_wait_task(&remote->task, -1);
```

dlech, 2022-11-26:
We are currently using GCC 10.3. But it is still good to fix things like this since we will eventually upgrade in a few years.

pbochynski, 2022-11-26:
I tried the casting you suggested but it has no effect. The compiled size of pb_type_pupdevices_Remote_light_on is still 0x50 bytes, and the brightness of green and blue is still zero.

dlech, 2022-11-26:
Maybe a compiler bug?

pbochynski, 2022-11-26:
Removing these 2 lines helps. Maybe yellow is not as yellow as it should be, but red is red, blue is blue, and green is green :)

pbochynski, 2022-11-28:
I couldn't get it fixed with ARM GNU tools 11.3. The compiler assumes that the payload contains 0 and doesn't try to do any operation on it. I downgraded to ARM GNU tools 10.3 and it works out of the box.

closed 2022-11-28 by pbochynski